esp-rs/esp-hal #868: Possible race condition in SpiDma transfers

bugadani, 2023-10-25:
https://github.com/esp-rs/esp-hal/blob/main/esp-hal-common/src/spi/master.rs#L1340

The DMA sets the `GDMA_OUT_EOF_CHn_INT_ENA` bit after it starts the transmission. This may result in the future never resolving, if some other code is executed between transmission start and listen start.

Or at least I think this is the case.

bjoernQ, 2023-10-25:
I think I've seen problem caused by this pattern with the AES peripheral which is faster (with small amounts of data) than we get to creating the future - so it's not just a hypothetical thing

bugadani, 2023-10-25:
I think I'm hitting this issue or something related with esp-wifi and a spidma-driven display; the display's DmaTxFuture often stops resolving if my mobile is connected to the device's AP.

PR #869 (2023-10-25): Fix async SPI/DMA race condition
## Thank you!

Thank you for your contribution.
Please make sure that your submission includes the following:

### Must

- [x] The code compiles without `errors` or `warnings`.
- [x] All examples work.
- [x] `cargo fmt` was run.
- [x] Your changes were added to the `CHANGELOG.md` in the proper section.
- [x] You updated existing examples or added examples (if applicable).
- [x] Added examples are checked in CI

### Nice to have

- [x] You add a description of your work to this PR.
- [x] You added proper docs for your newly added features and code.

Closes #868
 [... cut]
closed 2023-10-25 by MabezDev