esp-rs/esp-hal #870: Assistance with LifeTimes `embassy` + `ledc` example

Token-Thinker, 2023-10-25:
hello,

I think what I'm working on might be a useful example for the community, however i can't seem to get pass the lifetime issue when trying to convert the `ledc` example in `esp32-hal` so that i can use the `embassy executor` macro for task allocation.

Any hints would be instrumental in helping me with this process as i am a little stuck. 
Some solutions I've considered but didn't know if it was the best approach - "Creating a separate structure with `<'a>` for the `ledc` lifetime" and "potentially using `Mutex`". Both solutions seem a tad bit complicated for what i am trying to produce.

```rust
static EXECUTOR: StaticCell<Executor> = StaticCell::new();

#[entry]
fn main() -> ! {
    let peripherals = Peripherals::take();
    let mut system = peripherals.DPORT.split();
    let clocks = ClockControl::max(system.clock_control).freeze();

    let io = gpio::IO::new(peripherals.GPIO, peripherals.IO_MUX);
    let led= io.pins.gpio4.into_push_pull_output();

    let ledc = LEDC::new(peripherals.LEDC, &clocks, &mut system.peripheral_clock_control);
    
    let timer_group0 = TimerGroup::new(
        peripherals.TIMG0,
        &clocks,
        &mut system.peripheral_clock_control,
    );
    
    embassy::init(&clocks, timer_group0.timer0);

    let executor = EXECUTOR.init(Executor::new());

    logger::init_logger_from_env();
    log::info!("Logger is setup");

    executor.run(|spawner| {
        spawner.spawn(breathe(led,ledc)).ok();
    })
}

#[embassy_macros::task]
async fn breathe(led: GpioPin<Output<PushPull>, 4>, ledc: LEDC<'_>) {    
    let mut hstimer0 = ledc.get_timer::<HighSpeed>(timer::Number::Timer0);

    hstimer0
    .configure(timer::config::Config {
        duty: timer::config::Duty::Duty5Bit,
        clock_source: timer::HSClockSource::APBClk,
        frequency: 24u32.kHz(),
    })
    .unwrap();

    let mut channel0 = ledc.get_channel(channel::Number::Channel0, led);
    channel0
        .configure(channel::config::Config {
            timer: &hstimer0,
            duty_pct: 10,
            pin_config: channel::config::PinConfig::PushPull,
        })
        .unwrap();

    channel0.start_duty_fade(0, 100, 2000).expect_err(
        "Fading from 0% to 100%, at 24kHz and 5-bit resolution, over 2 seconds, should fail",
    );
    
    loop {
        // Set up a breathing LED: fade from off to on over a second, then
        // from on back off over the next second.  Then loop.
        channel0.start_duty_fade(0, 100, 1000).unwrap();
        while channel0.is_duty_fade_running() {}
        
        channel0.start_duty_fade(100, 0, 1000).unwrap();
        while channel0.is_duty_fade_running() {}
    }
}
```

bugadani, 2023-10-26:
Embassy tasks can't be generic and that includes lifetimes, too. You'll need to pass `LEDC<'static>` to it in some form.

bjoernQ, 2023-10-26:
You could take inspiration from https://github.com/esp-rs/esp-hal/blob/0064766ef2fd1ba6c51109cb75901e2122d5413a/esp32c3-hal/examples/embassy_spi.rs#L40-L51

Token-Thinker, 2023-10-27:
thank you both for your suggestions. 

@bugadani - the problem I'm having is if i just pass `LEDC<`static>` this causes an issue with the `&clocks` because that also would have to be passed as static as well. I tired messing around and doing different combinations but always get stuck with the LifeTime issue.

@bjoernQ - I tired importing the `static_cell` crate to utilize `makestatic!` similar to the example you showed, which theretically would work but that throws my build into wack because i'm using the toolchain `channel = "esp"` vs `nightly`. if you have a suggestion on how to use/fix that complie issue I believe using the `makestatic!` would be the route i would like to take.

thank you again for your patience.

ideally I should just have to pass the `channel0` variable like below and that could technically be constructed if i use the `makestatic!` if i am understanding the documentation correctly.

```rust
#[embassy_macros::task]
async fn breathe(channel0: hal::ledc::channel::Channel<'static, HighSpeed, GpioPin<Output<PushPull>, 4>> ) {        
    loop {
        // Set up a breathing LED: fade from off to on over a second, then
        // from on back off over the next second.  Then loop.
        channel0.start_duty_fade(0, 100, 1000).unwrap();
        while channel0.is_duty_fade_running() {}
        
        channel0.start_duty_fade(100, 0, 1000).unwrap();
        while channel0.is_duty_fade_running() {}
    }
}
```

bjoernQ, 2023-10-27:
Weird you get build errors with the "esp" toolchain - we used to use static_cell in the examples also for the Xtensa based chips.
(e.g. here https://github.com/esp-rs/esp-hal/blob/0064766ef2fd1ba6c51109cb75901e2122d5413a/esp32-hal/examples/embassy_spi.rs#L40-L51 )

Otherwise you could also transmute the lifetime in an unsafe block

Token-Thinker, 2023-10-29:
@bjoernQ - thank you for the reply, not really sure on how _transmute the lifetime in an unsafe block_ still relatively new with concepts so i went with a brute force method. It is working but ugly so any advice on a refactor as i would like to put it in a `struct` (like the one below as an example) is welcomed as i muddle along.

Note wasn't able to use `static_cell` instead i used the `embassy_executor::_export::StaticCell;`which I think had the same effect.

```rust
pub struct SystemInit<'a> {
    executor: Executor,
    clocks: Clocks<'a>,
    ledc: LEDC<'a>,
    hstimer0: hal::ledc::timer::Timer<'a, HighSpeed>,
}

static SYSTEM_INIT: StaticCell<SystemInit> = StaticCell::new();
```

```rust
static EXECUTOR: StaticCell<Executor> = StaticCell::new();
static CLOCKS: StaticCell<Clocks> = StaticCell::new();
static LEDC: StaticCell<LEDC> = StaticCell::new();
static HSTIMER0: StaticCell<hal::ledc::timer::Timer<'static, HighSpeed>> = StaticCell::new();

#[entry]
fn main() -> ! {
    let peripherals = Peripherals::take();
    let mut system = peripherals.DPORT.split();

    let clocks_control = ClockControl::max(system.clock_control).freeze();
    let clocks = CLOCKS.init(clocks_control);

    let ledc = LEDC.init_with(|| {
        LEDC::new(peripherals.LEDC, clocks, &mut system.peripheral_clock_control)
    });

    let io = gpio::IO::new(peripherals.GPIO, peripherals.IO_MUX);
    let led = io.pins.gpio4.into_push_pull_output();

    let hstimer0 = HSTIMER0.init_with( || {
        ledc.get_timer::<HighSpeed>(timer::Number::Timer0)
    });

    hstimer0
    .configure(timer::config::Config {
        duty: timer::config::Duty::Duty5Bit,
        clock_source: timer::HSClockSource::APBClk,
        frequency: 24u32.kHz(),
    })
    .unwrap();

    let mut channel0 = ledc.get_channel(channel::Number::Channel0, led);
    channel0
        .configure(channel::config::Config {
            timer: hstimer0,
            duty_pct: 10,
            pin_config: channel::config::PinConfig::PushPull,
        })
        .unwrap();

    channel0.start_duty_fade(0, 100, 2000).expect_err(
        "Fading from 0% to 100%, at 24kHz and 5-bit resolution, over 2 seconds, should fail",
    );

    let timer_group0 = TimerGroup::new(
        peripherals.TIMG0,
        &clocks,
        &mut system.peripheral_clock_control,
    );
    
    embassy::init(&clocks, timer_group0.timer0);

    let executor = EXECUTOR.init(Executor::new());

    logger::init_logger_from_env();
    log::info!("Logger is setup");

    executor.run(|spawner| {
        spawner.spawn(breathe(channel0)).ok();
    })
}

#[embassy_macros::task]
async fn breathe(channel0: hal::ledc::channel::Channel<'static, HighSpeed, GpioPin<Output<PushPull>, 4>> ) {        
    loop {
        channel0.start_duty_fade(0, 100, 1000).unwrap();
        while channel0.is_duty_fade_running() {}
        
        channel0.start_duty_fade(100, 0, 1000).unwrap();
        while channel0.is_duty_fade_running() {}
    }
}

```