esp-rs/esp-hal #872: Unify the low-power peripheral names and rename the `RtcCntl` driver

jessebraham, 2023-10-25:
At present, depending on which chip you're targeting, this driver is initialized in one of two ways:

```rust
// esp32c6 & esp32h2:
let rtc_cntl = RtcCntl::new(peripherals.LP_CLKRST);

// Everything else:
let rtc_cntl = RtcCntl::new(peripherals.RTC_CNTL);
```

There are two major issues with our current implementation:

1. The peripheral singleton structs should have their names unified; it's confusing passing two different peripherals to the same constructor, depending on which chip you're using. It's also confusing that for the C6 and H2 the peripheral name is seemingly unrelated to the driver's name.
2. `RtcCntl` is just not a good name. This was chosen initially as it was used in the TRMs, however with the new naming schemes this is less relevant now. This name is also, IMO, incredibly misleading, as may people associate it with Real-Time Clocks instead.

So the question remains, what should we call these things?

bugadani, 2023-10-25:
For me, anything goes as long as it's not RTC. Let me throw LP_CNTL in the pool of possibilities :)

bjoernQ, 2023-10-25:
I double that - anything not containing RTC is fine to me

jessebraham, 2023-10-30:
Is `LP_CLKRST` or even just `CLKRST` maybe not a bad name to use? It's at least consistent with _some_ chips this way. Just a thought.

bugadani, 2023-10-30:
I think LP_CNTL is broad enough so that it covers the power domain stuff, too, that I think belongs to RTC in the S3. CLKRST is somewhat more ... concrete and may not be precise enough because of that.

MabezDev, 2023-10-30:
Throwing my suggestion into the ring:

`LPWR` and `LowPowerDomain`

PR #1064 (2024-01-08): Unify the low-power peripheral names (`RTC_CNTL` and `LP_CLKRST` to `LPWR`)
closes #872
closed 2024-01-08 by jessebraham