esp-rs/esp-hal #881: esp32c3 floating-point related linking error

mchodzikiewicz, 2023-10-28:
Hi,

I noticed that making some operations on for example f32 can refer to symbols that are not existing for the target and outcome is a linking error.

more or less minimal example:
```
#[entry]
fn main() -> ! {
    let limit = core::hint::black_box(0f32);
    let value = core::hint::black_box(0f32);
    let f = limit.min(value);
    loop {}
}
```
results in
```
  = note: rust-lld: error: undefined symbol: fminf
          >>> referenced by f32.rs:929 (/usr/local/rustup/toolchains/nightly-x86_64-unknown-linux-gnu/lib/rustlib/src/rust/library/core/src/num/f32.rs:929)
          >>>               /builds/mchodzikiewicz/reproduce-esp32-c3-fminf/target/riscv32imc-unknown-none-elf/debug/deps/reproduce_esp32_c3_fminf-5e6a2dffd6f60119.24zm9j25iyrq5tis.rcgu.o:(core::f32::_$LT$impl$u20$f32$GT$::min::h8f6fefa25644f8e5)
```

Since this target does not have FPU, I tried to add   `-C soft-float` to rustflags but no change in behavior.

I have no idea where this issue belongs, but since I stumbled on it working on esp32c3 project, I guess it can be a good starting point.

example is available here: https://gitlab.com/mchodzikiewicz/reproduce-esp32-c3-fminf
and here you can see a reproduction: https://gitlab.com/mchodzikiewicz/reproduce-esp32-c3-fminf/-/jobs/5407620084

mchodzikiewicz, 2023-10-29:
It seems to be LLVM thing - https://github.com/llvm/llvm-project/issues/64206

MabezDev, 2023-10-30:
How old is your compiler? These intrinsics should be emitted by a recent version of rustc in compiler-builtins: https://github.com/rust-lang/compiler-builtins/blob/d1d785a538d51d0082c3e6209a28c9fb1a90ec3c/src/math.rs#L142-L157

mchodzikiewicz, 2023-10-30:
I created a docker container just to show the reproduction on saturday 28th.
```
#5 0.443 info: latest update on 2023-10-28, rust version 1.75.0-nightly (2f1bd0729 2023-10-27)
```
So it is not outdated for sure

MabezDev, 2023-10-30:
Ah, looks like a bug in compile-builtins, the math module isn't enabled for riscv without floating point: https://github.com/rust-lang/compiler-builtins/blob/d1d785a538d51d0082c3e6209a28c9fb1a90ec3c/src/lib.rs#L46-L56

mchodzikiewicz, 2023-10-30:
does this mean that the fix is as simple as
```diff
diff --git a/src/lib.rs b/src/lib.rs
index d1195a4..dd4ede7 100644
--- a/src/lib.rs
+++ b/src/lib.rs
@@ -48,6 +48,7 @@ pub mod int;
     all(target_arch = "x86_64", target_os = "none"),
     all(target_arch = "x86_64", target_os = "uefi"),
     all(target_arch = "arm", target_os = "none"),
+    all(target_arch = "riscv32", target_os = "none"),
     all(target_arch = "xtensa", target_os = "none"),
     all(target_arch = "mips", target_os = "none"),
     target_os = "xous",
```
or there's a reason for this not being enabled?

mchodzikiewicz, 2023-10-30:
sure, thanks! it was very helpful