esp-rs/esp-hal #882: UART2 data corruption

shyndman, 2023-10-29:
Hi there,

I've been encountering `rxfifo` data corruption while working with esp32s3's UART2 peripheral (UART0/1 work fine). I'm depending on 29ebd170eee686e61683ae704d7e9b4a834040a0.

The bytes I'm reading off the wire are not correct, as confirmed by a logic analyzer.

Looks like the problem is related to the hardware, as the [CircuitPython folks encountered the same problem](https://github.com/adafruit/circuitpython/issues/6254) awhile back. 

They resolved the issue by resetting `txfifo` and `rxfifo` buffers during peripheral setup, which appears to fix the issue with esp-rs as well.

Let me know if you need anything else to confirm the issue. Thanks!

elpiel, 2023-11-02:
I have a similar issue on the Rust ESP32-C3 board but I'm using `UART1`. I think I also tried `UART0`

bjoernQ, 2023-12-01:
I cannot reproduce this on my side. Do you have some minimal code to reproduce the issue?

MabezDev, 2023-12-01:
I believe this is more likely the issue described here: https://github.com/esp-rs/esp-hal/issues/989. In the case of FIFO overflow, the FIFO still isn't emptied meaning incoming data is lost. Closing this in favour of https://github.com/esp-rs/esp-hal/issues/989. A repro would still be nice just incase it isn't https://github.com/esp-rs/esp-hal/issues/989.

closed 2023-12-01 by MabezDev