esp-rs/esp-hal #884: Implement `Digital Signature` peripheral

AnthonyGrondin, 2023-10-30:
## Description
A Digital Signature is used to verify the authenticity and integrity of a message using a cryptographic algorithm.
This can be used to validate a device’s identity to a server, or to check the integrity of a message.

The Digital Signature (DS) module provides hardware acceleration of messages’ signatures
based on RSA. It uses pre-encrypted parameters to calculate a signature. The parameters are encrypted using
HMAC as a key-derivation function. In turn, the HMAC uses eFuses as an input key. The whole process happens
in hardware so that neither the decryption key for the RSA parameters nor the input key for the HMAC key
derivation function can be seen by the users while calculating the signature.

## Support
If a cell contains am em dash (—) this means that the particular peripheral is not present for a chip. A check mark (✓) means that some driver implementation exists. An empty cell (  ) means that the driver hasn't been implemented for the chip yet.

| ESP32 | ESP32-C2 | [ESP32-C3](https://www.espressif.com/sites/default/files/documentation/esp32-c3_technical_reference_manual_en.pdf#section.22) | [ESP32-C6](https://www.espressif.com/sites/default/files/documentation/esp32-c6_technical_reference_manual_en.pdf#section.23) | [ESP32-H2](https://www.espressif.com/sites/default/files/documentation/esp32-h2_technical_reference_manual_en.pdf#section.22) | [ESP32-S2](https://www.espressif.com/sites/default/files/documentation/esp32-s2_technical_reference_manual_en.pdf#section.20) | [ESP32-S3](https://www.espressif.com/sites/default/files/documentation/esp32-s3_technical_reference_manual_en.pdf#section.22) |
|-----------|---------------|----------------|----------------|----------------|---------------|----------------|
|     —    |        —       |                   |                   |                   |                  |                   |

## Additional information
Here's an example of how the peripheral is used inside `esp-idf` to accelerate signatures for `mbedtls`
https://github.com/espressif/esp-idf/blob/master/components/mbedtls/port/esp_ds/esp_rsa_sign_alt.c

brxken128, 2023-11-12:
With this functionality, these devices can implement some form of 2FA or act as a HSM which would be a huge bonus. I'm a cryptographer myself, and would be more than willing to give writing a Rust-compatible challenge-response library utlising these features.

I only have access to ESP32-S3's (as they're my favourite), but I'll pick up a few other suitable chips in order to validate things across more devices in due course.

chipweinberger, 2026-10-12:
looking forward to this!