esp-rs/esp-hal #890: Allow using FullDuplexMode SPI with any pin combinations (`mosi, miso, cs, ...`)

sanxus, 2023-11-01:
Hi!

I'm using a crate called `display-interface-spi` to drive my LCD module.
It needs trait bond `embedded_hal::blocking::spi::Write` to be satisfied.
```
impl<SPI, DC> SPIInterfaceNoCS<SPI, DC>
where
    SPI: hal::blocking::spi::Write<u8>,
    DC: OutputPin,
{
    pub fn new(spi: SPI, dc: DC) -> Self {
        Self { spi, dc }
    }
    
    // other methods...
}
```

I know I can call `Spi::new_half_duplex` if I want those pins to be optional, but it is *HalfDuplex* which not implement trait `embedded_hal::blocking::spi::Write`.
So, it seems like one can only use a instance created by `Spi::new` method which takes all these pins away, if that trait bond need to be satisfied?

I think there's two solutions:
 - make `Spi::new` that FullDuplexMode SPI can be used with optional pins
 - implement `embedded_hal::blocking::spi::Write` for HalfDuplexMode SPI

Maybe there's another way to do this thing?

bjoernQ, 2023-11-01:
For half-duplex we have a constructor taking Option for each pin. We probably should add a similar constructor for full-duplex.

However, for full-duplex we have a couple of constructors for the most common configurations:
- [new](https://docs.rs/esp32c3-hal/0.13.0/esp32c3_hal/spi/master/struct.Spi.html#method.new)
- [new_no_cs](https://docs.rs/esp32c3-hal/0.13.0/esp32c3_hal/spi/master/struct.Spi.html#method.new_no_cs)
- [new_no_miso](https://docs.rs/esp32c3-hal/0.13.0/esp32c3_hal/spi/master/struct.Spi.html#method.new_no_miso)
- [new_no_cs_no_miso](https://docs.rs/esp32c3-hal/0.13.0/esp32c3_hal/spi/master/struct.Spi.html#method.new_no_cs_no_miso)
- [new_mosi_only](https://docs.rs/esp32c3-hal/0.13.0/esp32c3_hal/spi/master/struct.Spi.html#method.new_mosi_only

sanxus, 2023-11-02:
But not all combinations of them.
What I want is one with `sck`, `mosi` and `cs`. (no `miso`)

The LCD module has no data output pin, and it's okay to use hardware controlled chip-select pin since it's the only device on the SPI bus.
I think it's better to use a hardware controlled `cs` when possible since it eliminates delay between spi transfers.

> We probably should add a similar constructor for full-duplex.

It's definitely good to have one.

sanxus, 2023-11-02:
Then what the name of this new constructor should be?
Since already have [Spi::new](https://docs.rs/esp32c3-hal/0.13.0/esp32c3_hal/spi/master/struct.Spi.html#method.new), I have no idea.

I'm willing to make a PR.

bjoernQ, 2023-11-02:
I think a breaking change (i.e. making the pins in `Spie::new` Option) is not too much of a problem given we currently don't guarantee API stability

bugadani, 2023-11-02:
I would like to add that a constructor with a lot of parameters isn't ideal UX. I regularly have to loop up the pin order for `I2C::new` and that only has two. If you are considering a breaking change, maybe this could be improved, too.

My first suggestion would be a builder-like API, where the initial call sets up common parameters, then `.with_mosi()`, `.with_miso()` etc. could be used to set up individual pins.

bjoernQ, 2023-11-02:
Ideally, we find a way to have a builder which doesn't require weird generic parameters if we leave out some of the pins - I think that was the reason we skipped builders initially - but I agree, mixing up the pins also happens to me

bugadani, 2023-11-02:
I imagine the optional pins can be initialized to [`NO_PIN`](https://docs.rs/esp32s3-hal/latest/esp32s3_hal/gpio/constant.NO_PIN.html) by default, though that's just paper engineering in the morning (i.e. it may not be so simple).

PR #901 (2023-11-08): Improve spi master constructor
## Thank you!

Thank you for your contribution.
Please make sure that your submission includes the following:

### Must

- [ ] The code compiles without `errors` or `warnings`.
- [ ] All examples work.
- [x] `cargo fmt` was run.
- [x] Your changes were added to the `CHANGELOG.md` in the proper section.
- [ ] You updated existing examples or added examples (if applicable).
- [ ] Added examples are checked in CI

### Nice to have

- [ ] You add a description of your work to this PR.
- [x] You added proper docs for your newly added features and code.

Fix #890

This [... cut]
closed 2023-11-08 by bjoernQ