esp-rs/esp-hal #902: Getting raw ADC values with real resolution

katyo, 2023-11-06:
My initial implementation of ADC with calibration allowed me get raw values with 12 bit resolution.
This values can be easily converted to millivolts when needed.

The PR #836 removed that useful feature.

MabezDev, 2023-11-07:
I didn't end up reviewing #836, so I don't have much context here, but the PR states that the calibration was being done in millivolts so the extra resolution was essentially lost already because the calibration would be wrong. I believe you wrote the original calibration code, so do you have any comments on that?

My 2c is that we should have an easy API for obtaining the raw 12bit resolution and the mv converted values; we don't need to pick just one.

elpiel, 2023-11-07:
Yes I agree that you would like to have both.
Of course calibrated values would be best suited for both cases.

jneem, 2023-11-20:
Isn't `AdcCalBasic` the API for obtaining the raw 12bit resolution? Whatever the units for `AdcCalLine` are, if the gain coefficient is less than 1 there will be a loss of resolution.

MabezDev, 2026-10-27:
Since #6360, `AdcCalBasic` returns the raw result at full resolution and `AdcCalLine`/`AdcCalCurve` return millivolts, which covers this request; please reopen if it does not: https://github.com/esp-rs/esp-hal/commit/c824e15305d16b833384ab5b8d5eb74bdceaf5c5

closed 2026-10-27 by MabezDev