esp-rs/esp-hal #908: Figure out if DMA interrupts can be set up automagically by driver code

bugadani, 2023-11-07:
It's easy to forget enabling a DMA channel, which in turn may e.g. prevent SpiDma from operating correctly. As a user, I would appreciate if the HAL would deal with this for me, while still providing me with a way to specifying priorities.

bjoernQ, 2023-11-08:
Probably something in the same direction: if we find a reasonable way for this, we also might want to have that for async drivers which need their interrupts enabled, too. Or at least warn / panic if they are not enabled. (But then we could also just enable them, I guess)

bugadani, 2023-11-08:
We can call `interrupt::enable` multiple times, right? Maybe the solution is simpler than I thought yesterday: we can enable DMA interrupts with highest priority, and the user can call `interrupt::enable` to change that if they need to.

closed 2023-12-12 by jessebraham