esp-rs/esp-hal #917: ESP32-S3 PSRAM Not Stable

raphaelhetzel, 2023-11-09:
Hi,
while experimenting with the MMU (tried mmaping flash) I noticed that the PSRAM on the ESP32-S3 does not predictably maintain the set value.
I would assume it is only maintained while it is in cache.

One can replicate the results by modifying the PSRAM example as shown below.
The example also seems to be broken in its default state (Exception occured 'LoadProhibited' when creating the string)
I tested this on two ESP32-S3s with *MBs of PSRAM.

Is this reproducible or did I misinterpret how it can be used?
Thanks for any hints.

```
    ...
    println!("Going to access PSRAM");

    unsafe {
        let start = psram::psram_vaddr_start() as *mut u8;
        for i in 0..psram::PSRAM_BYTES {
            *start.add(i) = 0xAB;
            if(*start.add(i) != 0xAB) {
                println!("failed set");
            }
        }
        for i in 0..psram::PSRAM_BYTES {
            if(*start.add(i) != 0xAB) {
                println!("failed check");
            }
        }
    }

    loop {}
```

bjoernQ, 2023-11-09:
What exact ESP32-S3 boards are that?

Background: there are different PSRAM chips and we don't support all of them. (Probably more a thing for older ESP32)
Might your PSRAM be octal-SPI-RAM?

If the example already fails unmodified then something is quite wrong

raphaelhetzel, 2023-11-09:
Ok, sorry for causing a fuss here, turns out I just misread the datasheet.
I was most testing on a https://www.lilygo.cc/products/t5-4-7-inch-e-paper-v2-3, which has a ESP32-S3-WROOM-1-N16R8m, which actually uses octal-SPI RAM. With the correct RAM type selected this issue does not not exist.

bjoernQ, 2023-11-10:
Thanks for confirmation! Great to hear it's working for you
I guess we can close this, then

closed 2023-11-10 by bjoernQ