esp-rs/esp-hal #925: RMT asynch does not allow for concurrent TX and RX

edgsousa, 2023-11-10:
The issue seems to be from the RMT interrupt handler in `esp_hal_common::rmt::asynch` clearing all the enabled interrupts instead of the channel being handled.

https://github.com/esp-rs/esp-hal/blob/07ed22df17f28f1b1890aa47c3ae599182397712/esp-hal-common/src/rmt.rs#L1127
https://github.com/esp-rs/esp-hal/blob/07ed22df17f28f1b1890aa47c3ae599182397712/esp-hal-common/src/rmt.rs#L1259

bjoernQ, 2023-11-13:
Thanks for the bug report and the detailed analysis.

You are right and the problem is caused by those lines. Those lines are there for no reason since the bits for the individual channels are already reset before - so the fix would be to just delete that code. They are most probably a left over from some testing.

Would you like to create a PR or should I create a PR for this fix?

bjoernQ, 2023-11-16:
I guess with 930 merged we can close this

closed 2023-11-16 by bjoernQ