esp-rs/esp-hal #940: Implement common interfaces with PARL_IO

bugadani, 2023-11-14:
I'd love to see adapters that turn PARL_IO into (maybe DMA-driven) SPI, UART and whatever else it can implement.

Dominaezzz, 2024-09-24:
Are you still interested in this btw? (I was attempting to read from a camera but I hit https://github.com/esp-rs/esp-hal/issues/1867).

I ask because I don't know if you'd want these sort of examples here in esp-hal or would prefer them in https://github.com/esp-rs/esp-hal-community .

bugadani, 2024-09-24:
I _almost_ forgot this. I wanted to write an SPI driver based on PARL_IO but never got to it. I think such additions would be ideal in one or more community crates.

MabezDev, 2026-10-27:
Adapters on top of `PARL_IO` belong in a community crate such as [esp-hal-community](https://github.com/esp-rs/esp-hal-community), so we close this as not planned.

closed 2026-10-27 by MabezDev