esp-rs/esp-hal #942: Implement rand traits for Rng

Natureshadow, 2023-11-14:
In order to use the hardware RNG with external code that accepts implementations of `RngCore` and `CryptoRng`, these traits should be implemented for the `Rng` struct.

Natureshadow, 2023-11-15:
Looking at [Espressif's documentation about the ESP32 RNG](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/random.html), I learnt that the system has one random number generator, that produces randomness of different quality depending on conditions:

* real random data when any of WiFi, Bluetooth or the special bootloader entropy source is enabled
* pseudo random data if not

Hence, the `CryptoRng` trait must only be usable if one of the above conditions is true. I wonder what would be the best way to do this?

* let the `CryptoRng` impl check the conditions, and return an `Err` if hardware-backed randomness cannot be guaranteed
* let the `CryptoRng` simply turn on the bootloader entropy source

bjoernQ, 2023-11-16:
I guess the first option is better since the behavior for the second option might be somewhat surprising. (We could have it in the docs but often people only look into the docs if something isn't working as expected)

jessebraham, 2024-01-29:
We have implemented the necessary trait(s) in #1122, however (as is already documented) we have not met the necessary preconditions to produce _true_ random numbers. Should we consider this issue closed regardless, or should we keep it open until this changes? I think it's fine to close, but just want to double-check with others.

CC @bjoernQ @MabezDev

Natureshadow, 2024-01-29:
> however (as is already documented) we have not met the necessary preconditions to produce _true_ random numbers. Should we consider this issue closed regardless, or should we keep it open until this changes? I think it's fine to close, but just want to double-check with others.

AIUI, that means the CryptoRng trait should not be implemented now, only Rng should be implemented now.

This issue should be closed IMHO, and a follow-up issue to implement CryptoRng should be opened.

closed 2024-02-08 by jessebraham