esp-rs/esp-hal #946: Allowing easy configuration of IRAM placement of device drivers like ESP-IDF

ProfFan, 2023-11-15:
From my experiments in #939 it seems that with the ESP32 I-cache miss is a microsecond-level (several thousand cycles) penalty. In ESP-IDF they have a `Kconfig` based mechanism that allows putting devices drivers in the IRAM. We should have the same ability in order to enable timing-sensitive applications.

bugadani, 2023-11-15:
The first idea is to mark the interesting functions with `#[cfg_attr(feature = "optimize-XXX-in-iram", ram)]`. This enables choosing which peripherals are to be placed in iram, at the cost of a few feature flags.

ProfFan, 2024-01-19:
Another idea is to have a configuration file just like in `esp-wifi`: https://github.com/esp-rs/esp-wifi/blob/main/esp-wifi/docs/tuning.md

bjoernQ, 2024-03-13:
Isn't this already merged for SPI in #1096 ? But probably this issue is about drivers in general which depends on / is blocked by #1111 since we don't want a feature per driver

MabezDev, 2024-09-19:
With #2156 we can now define any number of config options to do this for individual drivers.

Closing.

closed 2024-09-19 by MabezDev