esp-rs/esp-hal #950: 4 bytes of RTC memory zeroed on ESP32c3 deep-sleep wakeup

Natureshadow, 2023-11-16:
On an esp32c3 board, the first 4 bytes of

```rust
static mut _RTC_MEM_PROTECTIVE_BYTES: [u8; 8] = [0u8; 8];
#[ram(rtc_fast, uninitialized)]
static mut BEACON_PUBLIC_KEY: [u8; 32] = [0u8; 32];
```

are zeroed on wake up from deep sleep

I will follow up with a reproducer soon.

Natureshadow, 2023-11-16:
As a workaround, protecting the first 4 bytes with padding helps:

```rust
#[ram(rtc_fast, uninitialized)]
static mut _RTC_MEM_PROTECTIVE_BYTES: [u8; 8] = [0u8; 8];
#[ram(rtc_fast, uninitialized)]
static mut BEACON_PUBLIC_KEY: [u8; 32] = [0u8; 32];
```

@bjoernQ Contrary to what you proposed in chat, setting the protective array to `zeroed` does not prevent the real data from being overwritten.

bjoernQ, 2023-11-17:
Simplest way to reproduce: use the `sleep_timer.rs` example, add this
```rust
#[ram(rtc_fast, uninitialized)]
static mut SOME_UNINITED_DATA: [u8; 20] = [0; 20];
```

and this (e.g. before `Delay::new`)
```rust
    unsafe {
        println!("data {:?}", SOME_UNINITED_DATA);

        SOME_UNINITED_DATA[0] = 42;
        SOME_UNINITED_DATA[1] = 23;    
        SOME_UNINITED_DATA[2] = 42;
        SOME_UNINITED_DATA[3] = 23;    
        SOME_UNINITED_DATA[4] = 42;
        SOME_UNINITED_DATA[5] = 23;    
        SOME_UNINITED_DATA[6] = 42;
        SOME_UNINITED_DATA[7] = 23;    
        SOME_UNINITED_DATA[8] = 42;
        SOME_UNINITED_DATA[9] = 23;    
    }
```

Workaround: Add this
```rust
#[ram(rtc_fast, zeroed)]
static mut SOME_ZEROED_DATA: [u8; 8] = [0; 8];
...
println!("zero {:?}", SOME_ZEROED_DATA);
```

It's caused by https://github.com/esp-rs/esp-hal/blob/c61287ec4f4f00087f2e38e9f344f9be969267e4/esp-riscv-rt/src/lib.rs#L304-L313

All the zero/copy routines should check the end-condition before copying/zeroing the first word

closed 2023-11-17 by MabezDev

Natureshadow, 2023-11-17:
@bjoernQ Thanks for fixing!

I am not yet sure whether it fixes the issue I see, because as laid out above, the workaround you concluded (with `zeroed`) doesn't prevent the issue here… but I will try.

PR #952 (2023-11-17): Fix zero/copy data in esp-riscv-rt
Fixes #950