esp-rs/esp-hal #954: DMA-driven SPI silently fails to transfer if data is in DCache

bugadani, 2023-11-17:
The epd-waveshare driver (or at least [this](https://github.com/caemor/epd-waveshare.git) fork) tends to send 1-4 bytes of data from DCache. The ESP32-S3 TRM states that accessing DCache is possible, but with restrictions:

> Any transmit and receive channels of GDMA can access 0x3C000000 ~ 0x3DFFFFFF in external RAM. GDMA
can send data only in burst mode. The number of data bytes to transfer in one burst is defined as block size.
Block size can be 16 bytes, 32 bytes or 64 bytes, configured via GDMA_IN_EXT_MEM_BK_SIZE_CHn for
transmit channels and GDMA_OUT_EXT_MEM_BK_SIZE_CHn for receive channels.

The epd-waveshare driver obviously doesn't meet the block size limit, and it doesn't matter if the application configures DMA for burst operation or not. The result is the same: nothing visible happens on the display.

IMO the driver should detect this situation and return an error.

MabezDev, 2023-11-17:
I went looking for how esp-idf handles it. It seems its possible to tell synchronize the buffer in cache with the DMA. They call [esp-cache_msync](https://github.com/espressif/esp-idf/blob/c8243465e45489835d645bf217a6929fd0c01b7f/components/driver/spi/gpspi/spi_master.c#L911-L914) in the driver before starting the transaction. The definition lives here: https://github.com/espressif/esp-idf/blob/c8243465e45489835d645bf217a6929fd0c01b7f/components/esp_mm/esp_cache.c#L29-L90.

MabezDev, 2023-11-17:
Huh, last time I went looking into how the cache works it was all ROM code and zero docs. Now there is a esp_mm component: https://github.com/espressif/esp-idf/tree/c8243465e45489835d645bf217a6929fd0c01b7f/components/esp_mm with some docs as to how it actually works. Maybe we should try and port some of this into esp-hal.

bugadani, 2023-11-17:
I'm not sure the added complexity is worth it for us at this point. The implementation would still just return an error for unaligned data address and/or size, which we can just as easily do for all data living in DCache. Though I'm not entirely sure how simple it is to align data in the cache, it seems more restrictive than telling the user that they should use the FlashSafeDma wrapper.

bjoernQ, 2024-03-11:
Somewhat related #1235

closed 2024-06-11 by JurajSadel