esp-rs/esp-hal #955: Cache/Memory map management

MabezDev, 2023-11-17:
Copied from https://github.com/esp-rs/esp-hal/issues/954#issuecomment-1816696280
> Huh, last time I went looking into how the cache works it was all ROM code and zero docs. Now there is a esp_mm component: https://github.com/espressif/esp-idf/tree/c8243465e45489835d645bf217a6929fd0c01b7f/components/esp_mm with some docs as to how it works. Maybe we should try and port some of this into esp-hal.

We could replace [this piece of code](https://github.com/esp-rs/esp-hal/blob/4d87e75d71b55d546b37aff6a4494cefb743c4a9/esp32s3-hal/src/lib.rs#L76-L100) (which is currently hardcoded) with a proper cache driver.

One limitation we have is that I don't think any of the cache registers are in the SVDs - but it looks like there is only one register with just a few bits: https://github.com/espressif/esp-idf/blob/c8243465e45489835d645bf217a6929fd0c01b7f/components/esp_mm/include/esp_cache.h#L21-L47

bjoernQ, 2023-11-17:
Ah we actually configure something for S3, didn't knew that 🤔 Oh - that was me

ProfFan, 2023-11-21:
> Ah we actually configure something for S3, didn't knew that 🤔 Oh - that was me

Is the DCACHE only enabled when using the PSRAM driver?

bjoernQ, 2023-11-27:
I _think_ for most targets the second stage bootloader will enable it and use whatever was configured when building it
For S3 (something I re-learned) we always configure it in esp-hal. It's a bit inconsistent IMHO

MabezDev, 2024-10-08:
I don't think this will be a priority for a long long while, closing for now.

closed 2024-10-08 by MabezDev

MarSik, 2025-04-24:
One question. I am working with DMA that sources data from PSRAM. According to the ESP docs the cpu access uses case when writing data to PSRAM, but DMA accesses the PSRAM directly. So I think (and my LCD agrees with me as pieces of the picture are missing) that I am missing a Cache to PSRAM sync.

The IDF provides https://docs.espressif.com/projects/esp-idf/en/stable/esp32s3/api-reference/system/mm_sync.html#memory-synchronization-driver , but sadly, I see no equivalent in esp-rs.

Did I miss anything or is there a gap in the functionality?

Dominaezzz, 2025-04-24:
https://github.com/esp-rs/esp-hal/blob/8f717c49bb16baa028ad8d99bffdd2985e336088/esp-hal/src/dma/buffers.rs#L626

MarSik, 2025-04-24:
Ah! Indeed. Sadly I am using my own implementation of the DmaTxBuffer and the whole `soc` is private. It is trivial though as it just calls few other rom functions. Thanks for the pointer.