esp-rs/esp-hal #956: Can't build async anymore

LarryMerino, 2023-11-18:
I had this base project that previously compiled without problems but I deleted the target folder and the Cargo.lock file and since then I have not been able to compile it again

```Toml
[dependencies]
hal = { package = "esp32c3-hal", version = "0.13.0", features = ["embassy-time-systick", "embassy", "eh1"]}
embedded-hal = { version = "=1.0.0-rc.1" }
esp-backtrace = { version = "0.9.0", features = ["esp32c3", "panic-handler", "exception-handler", "print-uart"] }
esp-println = { version = "0.7.0", features = ["esp32c3","log"] }
log = { version = "0.4.18" }
embassy-executor = { version = "0.3.2", features = [ "log", "nightly", "arch-riscv32", "executor-thread" ]}
embassy-time = { version = "0.1.5" }
embedded-hal-async = "1.0.0-rc.1"
```

I assume from the circumstances that it is some dependency problem but I have not been able to locate the source of the problem.

```Rust
#![no_std]
#![no_main]
#![feature(type_alias_impl_trait)]

use embassy_executor::Spawner;
use embedded_hal::spi::SpiDevice;
use esp_backtrace as _;
use esp_println::println;
use hal::{
    clock::ClockControl, 
    peripherals::Peripherals, 
    prelude::*,
    spi::{
        master::{prelude::*, Spi, SpiBusController},
        SpiMode,
    },
    Delay, 
    embassy,
    gpio::IO
};

#[main]
async fn main(_spawner: Spawner) -> ! {
    let peripherals = Peripherals::take();
    let system = peripherals.SYSTEM.split();

    let clocks = ClockControl::max(system.clock_control).freeze();
    let mut delay = Delay::new(&clocks);

    embassy::init(
        &clocks,
        hal::systimer::SystemTimer::new(peripherals.SYSTIMER),
    );

    let io = IO::new(peripherals.GPIO, peripherals.IO_MUX);

    let sclk = io.pins.gpio6;
    let miso = io.pins.gpio2;
    let mosi = io.pins.gpio7;

    let spi_controller = SpiBusController::from_spi(Spi::new_no_cs(
        peripherals.SPI2,
        sclk,
        mosi,
        miso,
        1000u32.kHz(),
        SpiMode::Mode0,
        &clocks,
    ));

    let can_device_1 = spi_controller.add_device(io.pins.gpio3);
    let can_device_2 = spi_controller.add_device(io.pins.gpio4);

[... 553 lines cut ...]
error[E0282]: type annotations needed
    --> /Users/larry/.cargo/registry/src/index.crates.io-6f17d22bba15001f/portable-atomic-1.5.1/src/lib.rs:2238:13
     |
2238 |             self.as_atomic_usize().bit_set(bit, order)
     |             ^^^^ cannot infer type for type parameter `T`

error[E0282]: type annotations needed
    --> /Users/larry/.cargo/registry/src/index.crates.io-6f17d22bba15001f/portable-atomic-1.5.1/src/lib.rs:2283:13
     |
2283 |             self.as_atomic_usize().bit_clear(bit, order)
     |             ^^^^ cannot infer type for type parameter `T`

error[E0282]: type annotations needed
    --> /Users/larry/.cargo/registry/src/index.crates.io-6f17d22bba15001f/portable-atomic-1.5.1/src/lib.rs:2328:13
     |
2328 |             self.as_atomic_usize().bit_toggle(bit, order)
     |             ^^^^ cannot infer type for type parameter `T`

Some errors have detailed explanations: E0282, E0412, E0433.
For more information about an error, try `rustc --explain E0282`.
error: could not compile `portable-atomic` (lib) due to 34 previous errors
warning: build failed, waiting for other jobs to finish...
```

Any ideas?

bugadani, 2023-11-18:
This isn't about async, this is about portable-atomic which does not currently build with esp-hal and risc-v CPUs. The [fix/workaround](https://github.com/esp-rs/esp-hal/pull/904) has been merged already.

LarryMerino, 2023-11-19:
Thank you very much for your answer @bugadani 

I'm trying to use the commit that references this fix 
```Toml
[dependencies]
hal = { git = "https://github.com/esp-rs/esp-hal.git", rev = "280caad3786bf3bb7c2de49d4aef8389b7be62db", package = "esp32c3-hal", version = "0.13.0", features = ["embassy-time-systick", "embassy", "eh1"]}
embedded-hal = { version = "=1.0.0-rc.1" }
esp-backtrace = { version = "0.9.0", features = ["esp32c3", "panic-handler", "exception-handler", "print-uart"] }
esp-println = { version = "0.7.0", features = ["esp32c3","log"] }
log = { version = "0.4.18" }
embassy-executor = { version = "0.3.2", features = [ "log", "nightly", "arch-riscv32", "executor-thread" ]}
embassy-time = { version = "0.1.5" }
embedded-hal-async = "1.0.0-rc.1"
```

But I can't get it to work

```
   Compiling embassy-executor v0.3.3
   Compiling esp32c3-hal v0.13.0 (https://github.com/esp-rs/esp-hal.git?rev=280caad3786bf3bb7c2de49d4aef8389b7be62db#280caad3)
   Compiling heapless v0.7.16
   Compiling quote v1.0.33
   Compiling litrs v0.4.1
   Compiling syn v2.0.39
   Compiling toml_edit v0.20.7
   Compiling proc-macro-crate v2.0.0
   Compiling darling_core v0.20.3
   Compiling serde_derive v1.0.192
   Compiling strum_macros v0.25.3
   Compiling darling_macro v0.20.3
   Compiling riscv-rt-macros v0.2.0
   Compiling darling v0.20.3
   Compiling esp-hal-procmacros v0.7.0 (https://github.com/esp-rs/esp-hal.git?rev=280caad3786bf3bb7c2de49d4aef8389b7be62db#280caad3)
   Compiling embassy-macros v0.2.1
   Compiling rustc-std-workspace-core v1.99.0 (/Users/larry/.rustup/toolchains/nightly-x86_64-apple-darwin/lib/rustlib/src/rust/library/rustc-std-workspace-core)
   Compiling critical-section v1.1.2
   Compiling nb v1.1.0
   Compiling void v1.0.2
   Compiling stable_deref_trait v1.2.0
   Compiling embedded-hal v1.0.0-rc.1
   Compiling byteorder v1.5.0
   Compiling pin-utils v0.1.0
   Compiling nb v0.1.3
   Compiling pin-project-lite v0.2.13
   Compiling embedded-hal v0.2.7
   Compiling bit_field v0.10.2
   Compiling embedded-hal-async v1.0.0-rc.1
   Compiling vcell v0.1.3
   Compiling hash32 v0.2.1
   Compiling cfg-if v1.0.0
   Compiling log v0.4.20
   Compiling gcd v2.3.0
   Compiling embedded-hal-nb v1.0.0-rc.1
   Compiling fugit v0.3.7
   Compiling embedded-can v0.4.1
   Compiling embedded-dma v0.2.0
   Compiling bitfield v0.14.0
   Compiling embedded-io v0.6.1
   Compiling strum v0.25.0
   Compiling bitflags v2.4.1
   Compiling esp-riscv-rt v0.5.0 (https://github.com/esp-rs/esp-hal.git?rev=280caad3786bf3bb7c2de49d4aef8389b7be62db#280caad3)
error[E0433]: failed to resolve: could not find `AtomicU8` in `imp`
   --> /Users/larry/.cargo/registry/src/index.crates.io-6f17d22bba15001f/portable-atomic-1.5.1/src/lib.rs:659:14
    |
659 |         imp::AtomicU8::is_lock_free()
    |              ^^^^^^^^ could not find `AtomicU8` in `imp`
    |
help: a struct with a similar name exists
    |
659 |         imp::AtomicU128::is_lock_free()
[... 522 lines cut ...]

error[E0282]: type annotations needed
    --> /Users/larry/.cargo/registry/src/index.crates.io-6f17d22bba15001f/portable-atomic-1.5.1/src/lib.rs:2238:13
     |
2238 |             self.as_atomic_usize().bit_set(bit, order)
     |             ^^^^ cannot infer type for type parameter `T`

error[E0282]: type annotations needed
    --> /Users/larry/.cargo/registry/src/index.crates.io-6f17d22bba15001f/portable-atomic-1.5.1/src/lib.rs:2283:13
     |
2283 |             self.as_atomic_usize().bit_clear(bit, order)
     |             ^^^^ cannot infer type for type parameter `T`

error[E0282]: type annotations needed
    --> /Users/larry/.cargo/registry/src/index.crates.io-6f17d22bba15001f/portable-atomic-1.5.1/src/lib.rs:2328:13
     |
2328 |             self.as_atomic_usize().bit_toggle(bit, order)
     |             ^^^^ cannot infer type for type parameter `T`

Some errors have detailed explanations: E0282, E0412, E0433.
For more information about an error, try `rustc --explain E0282`.
error: could not compile `portable-atomic` (lib) due to 34 previous errors
```

any suggestion?

JurajSadel, 2023-11-20:
Hello @LarryMerino! What does your `.cargo/config.toml` look like? Did you remove [these lines](https://github.com/esp-rs/esp-hal/pull/904/files#diff-393b621926aa438467c670f30f7261b68e7ad78beb471830b1fd576450a6fc7d)? I _think_ this might be the problem :)

LarryMerino, 2023-11-20:
Hi @JurajSadel 👋🏽
You're right, with this configuration it compiles correctly.
Thank you so much

closed 2023-11-20 by LarryMerino