esp-rs/esp-hal #965: Figure out if portable-atomic and the trap handler can coexist

MabezDev, 2023-11-20:
Before we cut a release, and possibly break a bunch of code, we should figure out if these are actually incompatible. It would suck to break the world and then unbreak it in the next release.

bugadani, 2023-11-20:
A thing to try might be codegenning for `-imac`, but I believe that is also something the users will need to do when updating.

bugadani, 2023-11-30:
[This PR](https://github.com/rust-lang/rust/pull/114499) lands in tomorrow's nightly, which may make some difference.

MabezDev, 2023-11-30:
I got tired of waiting for it in nightly and built the compiler myself. I don't think its possible for them to coexist nicely.

I think with https://github.com/rust-lang/rust/pull/114499 merged, we should probably just remove the trap handler entirely (xtensa too). The whole reason for the trap handler was because we couldn't just have load_store atomics without enabled `+a` extension in LLVM, and in doing that we had to then handle CAS operations. This was 2 years ago, things have moved on and portable-atomic is a really nice drop in replacement (and sometimes improvement!) over core.

I think I will make a PR to tear out the trap handler, and enable portable-atomic by default. The only downside to this of course is that we will be stuck on nightly until 1.76. I'd like to hear some input on how we should handle this.

bugadani, 2023-11-30:
> (xtensa too)

FYI that will break all heapless 0.7-based code as some stuff there uses compare-and-swap operations but does not polyfill them for Xtensa.

MabezDev, 2023-11-30:
I think the simple answer is update to 0.8 heapless :D, but good point, definitely worth noting that.

bugadani, 2023-11-30:
> Can't think of any other dependency

smoltcp 0.10 still uses it, along with embedded-nal-async