esp-rs/esp-hal #1028: Light-/deep-sleep support for ESP32-C6

jessebraham, 2023-12-14:
(Will add additional information shortly)

jessebraham, 2024-01-03:
#918 has been merged. I know this isn't feature-complete yet, @bugadani are you able to summarize what's left to do? Or should we just close this issue now?

bugadani, 2024-01-03:
There are two main areas that I ignored so far:
 - wakeup sources other than EXT1
 - various system state save/restore for light sleep

bjoernQ, 2024-07-01:
We now have working basic light-sleep and a few more wakeup-sources. I _think_ for remaining features we should have separate issues as needed instead of keeping this issue open forever

bjoernQ, 2024-07-04:
I suggest closing this issue and creating new issues as needed for retaining CPU state etc. when we see a need for it. Feel free to re-open if you disagree

closed 2024-07-04 by bjoernQ