esp-rs/esp-hal #1032: Direct Boot, please

perlindgren, 2023-12-14:
Hi Folks

Flashing the esp32c3 used to be super fast allowing for a nice workflow. Seems that the current flashing procedure introduces substantial overhead, and is thus a severe regression.

Can we please have the "direct-boot" back. If it considered unsafe, let's rename it to "unsafe-boot" or something.

/Per

MabezDev, 2023-12-15:
It's not unsafe, we removed it as they became too difficult to maintain. You can see the discussion we had in #803.

From my perspective, I think it would be better to improve the flashing speeds in espflash/probe-rs (by the way, the probe-rs master has had significant improvements) than support direct boot, but if concerns are addressed we could see direct-boot come back.

perlindgren, 2023-12-15:
Unsafe as in not supporting secure boot, but I guess its not supported yet in any case, so not any less secure than non-direct-boot then.

We saw there was an option to provide your own linker script, we will try that. (For small binaries, the difference in flash speed is huge, even when using latest probe-rs release). Moreover, it seems that non-direct-boot does not play well with probe-rs-debug + vscode plugin, but it may be just some missing option in the launch config. (It is not able to flash at all using the vscode plugin, but using `probe-rs run` it is possible, so it must be some flag propagation missing (--format idf). Similar story with `cargo embed`.

jessebraham, 2023-12-15:
Honestly, I really have no desire to deal with direct boot anymore, as @MabezDev alluded to it was a constant source of problems. It often got missed when updating linker scripts, and on multiple occasions was broken for weeks or months before anybody actually noticed.

If you insist on using this image format, I will say that you should probably provide your own linker scripts. If somebody wants to commit to maintaining this feature I would _potentially_ reconsider, but this would involve adding a `CODEOWNERS` file and I would expect that person to be on top of maintenance, and be willing to test/verify things before each release. I don't think this is a very reasonable expectation for a community member, but maybe somebody really wants this feature.

I agree that improving tooling would probably be a much better use of time than re-adding direct boot support to the HAL.

BestClaws, 2023-12-18:
> (For small binaries, the difference in flash speed is huge, even when using latest probe-rs release). 

have you tried the master branch, because last release i see was on october.

> (It is not able to flash at all using the vscode plugin, but using probe-rs run it is possible, so it must be some flag propagation missing (--format idf). Similar story with cargo embed

vscode does have this feature!

"flashingConfig": {
             ...
             "formatOptions": {
                 "format": "idf"
             }
         }

cargo-embed doesn't have this flag, but you can easily fork and change a single line to enable this, as cargo-embed is just a thin wrapper over probe-rs

perlindgren, 2023-12-18:
I figured there should be such an option, but I failed to get it working with the released version. I'll try installing from source and see if that works.

Thanks for the heads up by the way!

closed 2024-01-04 by MabezDev