esp-rs/esp-hal #1034: Remove the `embedded-hal@0.2.x` trait implementations when able

jessebraham, 2023-12-15:
I think, once we're able, we should rip out the old trait implementations and remove the `eh1` feature, implementing the latest stable traits by default.

Prerequisites:

- [ ] Stable `1.0.0` releases of the relevant `embedded-hal-*` packages
- [ ] [embedded-hal-compat](https://github.com/ryankurte/embedded-hal-compat) updated to use stable releases

bjoernQ, 2023-12-15:
Would definitely make things more maintainable. I think there are some traits which didn't made it into 1.0 e.g. Adc but that one has its issues so we probably should just provide our own (better) implementation

jessebraham, 2023-12-15:
> I think there are some traits which didn't made it into 1.0 e.g. Adc but that one has its issues so we probably should just provide our own (better) implementation

Yeah I think for the most part, the traits that do not have analogues in the `1.0.0` release are not use heavily (or at all) by driver code, so I'm not sure we're losing out on much there. In general I think we need to improve our APIs when not using trait methods anyway, so this will likely push us to do so for at least a few peripherals.

MabezDev, 2024-02-19:
I think that maybe removing 0.2 impls isn't the best idea, for one embedded-hal-compat can be quite finicky to work with. Secondly, https://github.com/ryankurte/embedded-hal-compat/pull/32#issuecomment-1952591174 would suck.

jessebraham, 2024-02-20:
> I think that maybe removing 0.2 impls isn't the best idea, for one embedded-hal-compat can be quite finicky to work with. Secondly, [ryankurte/embedded-hal-compat#32 (comment)](https://github.com/ryankurte/embedded-hal-compat/pull/32#issuecomment-1952591174) would suck.

Are you suggesting we keep the `0.2.x` and `1.0.0` implementations _forever_?

MabezDev, 2024-02-20:
Maybe not forever, but until the majority of the ecosystem drivers is on eh1. I think it's worth keeping around for now at least.

closed 2024-02-20 by jessebraham