esp-rs/esp-hal #1045: How to use EEPROM？

Song-aff, 2023-12-20:


bjoernQ, 2023-12-20:
It's not clear to me what you are asking for.

Do you talk about the EEPROM crate? Then you might want to look into esp-storage as an ESP32 alternative

Are you talking about some specific EEPROM chip? Which one? And which ESP32 you want to use?

Song-aff, 2023-12-20:
> It's not clear to me what you are asking for.
> 
> Do you talk about the EEPROM crate? Then you might want to look into esp-storage as an ESP32 alternative
> 
> Are you talking about some specific EEPROM chip? Which one? And which ESP32 you want to use?

Sorry, I would like to use some data that will not be lost after power failure to store configurations that can be set externally. Does ESP32 have such a built-in feature.

bjoernQ, 2023-12-20:
For persistence you can use https://github.com/esp-rs/esp-storage to write and read the flash. You should use a dedicated partition for the data. Our default partition table (defined in espflash) contains a NVS partition at flash address 0x9000

Song-aff, 2023-12-20:
Thank you for your answer. I will learn it

DBLouis, 2024-07-24:
> Our default partition table (defined in espflash) contains a NVS partition at flash address 0x9000

Could you please point where this is defined?

DBLouis, 2025-02-07:
> For persistence you can use https://github.com/esp-rs/esp-storage to write and read the flash. You should use a dedicated partition for the data. Our default partition table (defined in espflash) contains a NVS partition at flash address 0x9000

Is this NVS partiton at address 0x9000 used for other purpose? Because espflash says it is "WiFi data":
```
I (60) boot: Partition Table:
I (63) boot: ## Label            Usage          Type ST Offset   Length
I (71) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (78) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (86) boot:  2 factory          factory app      00 00 00010000 003f0000
I (93) boot: End of partition table
```