esp-rs/esp-hal #1046: Compilation failed, with the keyword ‘AtomicUsize‘.

khsdfyd, 2023-12-20:
I set up a project and enabled the following features, but it cannot be compiled.
`Cargo.toml`
```
[dependencies]
hal = { git = "https://github.com/esp-rs/esp-hal.git", package = "esp32c3-hal", version = "0.14.0"}
esp-backtrace = { version = "0.9.0", features = ["esp32c3", "panic-handler", "exception-handler", "print-uart"] }
esp-println = { version = "0.7.0", features = ["esp32c3","log"] }
log = { version = "0.4.18" }
esp-alloc = { version = "0.3.0" }
esp-wifi  = { version = "0.1.1", features = ["esp32c3", "wifi", "ble", "esp-now"] }
smoltcp = { version = "0.10.0", default-features=false, features = ["proto-igmp", "proto-ipv4", "socket-tcp", "socket-icmp", "socket-udp", "medium-ethernet", "proto-dhcpv4", "socket-raw", "socket-dhcpv4"] }
embedded-svc = { version = "0.25.0", default-features = false, features = [] }
embedded-io = "0.4.0"
heapless = { version = "0.7.14", default-features = false }

embassy-executor = { version = "0.4.0", features = ["nightly", "integrated-timers", "arch-riscv32", "executor-thread"] }
embassy-time   = { version = "0.2.0", optional = true }
critical-section   = "1.1.2"
embassy-sync       = "0.4.0"
static_cell        = { version = "2.0.0", features = ["nightly"] }

[profile.dev.package.esp-wifi]
opt-level = 3
```
`.cargo/config.toml`
```
[target.riscv32imc-unknown-none-elf]
runner = "espflash flash --monitor"
rustflags = [
  "-C", "link-arg=-Tlinkall.x",
  "-C", "force-frame-pointers",
]
[env]
ESP_LOGLEVEL="INFO"
[build]
target = "riscv32imc-unknown-none-elf"
[unstable]
build-std = ["alloc", "core"]
```
`src/main.rs`
```
#![no_std]
#![no_main]

extern crate alloc;
use core::mem::MaybeUninit;
use esp_backtrace as _;
use esp_println::println;
use hal::{clock::ClockControl, peripherals::Peripherals, prelude::*, Delay};

use esp_wifi::{initialize, EspWifiInitFor, current_millis};
use esp_wifi::esp_now::{PeerInfo, BROADCAST_ADDRESS};

use hal::{systimer::SystemTimer, Rng};
#[global_allocator]
static ALLOCATOR: esp_alloc::EspHeap = esp_alloc::EspHeap::empty();

fn init_heap() {
    const HEAP_SIZE: usize = 32 * 1024;
    static mut HEAP: MaybeUninit<[u8; HEAP_SIZE]> = MaybeUninit::uninit();

    unsafe {
        ALLOCATOR.init(HEAP.as_mut_ptr() as *mut u8, HEAP_SIZE);
    }
}

#[entry]
fn main() -> ! {
    init_heap();
    let peripherals = Peripherals::take();
    let system = peripherals.SYSTEM.split();
    let clocks = ClockControl::max(system.clock_control).freeze();
    let mut delay = Delay::new(&clocks);
    
    // esp_println
    // setup logger
    // To change the log_level change the env section in .cargo/config.toml
    // or remove it and set ESP_LOGLEVEL manually before running cargo run
    // this requires a clean rebuild because of https://github.com/rust-lang/cargo/issues/10358
    esp_println::logger::init_logger(log::LevelFilter::Info);

    // esp_println::logger::init_logger_from_env();
    log::info!("Logger is setup");
    println!("Hello world!");
    let timer = SystemTimer::new(peripherals.SYSTIMER).alarm0;
    let init = initialize(
        EspWifiInitFor::Wifi,
        timer,
        Rng::new(peripherals.RNG),
        system.radio_clock_control,
        &clocks,
    )
    .unwrap();
    // esp_wifi::

    let wifi = peripherals.WIFI;
    let mut esp_now = esp_wifi::esp_now::EspNow::new(&init, wifi).unwrap();
    println!("esp-now version {}", esp_now.get_version().unwrap());

    let mut next_send_time = current_millis() + 5 * 1000;
    loop {
        let r = esp_now.receive();
        if let Some(r) = r {
            println!("Received {:?}", r);

            if r.info.dst_address == BROADCAST_ADDRESS {
                if !esp_now.peer_exists(&r.info.src_address) {
                    esp_now
                        .add_peer(PeerInfo {
                            peer_address: r.info.src_address,
                            lmk: None,
                            channel: None,
                            encrypt: false,
                        })
                        .unwrap();
                }
                let status = esp_now
                    .send(&r.info.src_address, b"Hello Peer")
                    .unwrap()
                    .wait();
                println!("Send hello to peer status: {:?}", status);
            }
        }

        if current_millis() >= next_send_time {
            next_send_time = current_millis() + 5 * 1000;
            println!("Send");
            let status = esp_now
                .send(&BROADCAST_ADDRESS, b"0123456789")
                .unwrap()
                .wait();
            println!("Send broadcast status: {:?}", status)
        }
    }
}

```
I tried [this,](https://github.com/esp-rs/esp-hal/issues/956) but it was not successful.

khsdfyd, 2023-12-20:
This is the output
```
   Compiling futures-task v0.3.29
   Compiling futures-core v0.3.29
   Compiling embassy-time v0.2.0
   Compiling embedded-hal-async v1.0.0-rc.2
   Compiling darling v0.20.3
   Compiling basic-toml v0.1.7
   Compiling toml v0.5.11
   Compiling futures-util v0.3.29
   Compiling atomic-waker v1.1.2
   Compiling embedded-hal-nb v1.0.0-rc.2
   Compiling esp-wifi-sys v0.1.0
   Compiling num-traits v0.2.17
   Compiling esp-println v0.7.1
   Compiling libm v0.2.8
   Compiling num-derive v0.3.3
   Compiling atomic_enum v0.2.0
   Compiling embedded-io v0.4.0
   Compiling esp-alloc v0.3.0
error[E0599]: no method named `compare_exchange` found for struct `AtomicUsize` in the current scope
   --> C:\Users\khsdfyd\.cargo\registry\src\rsproxy.cn-0dccff568467c15b\atomic-waker-1.1.2\src\lib.rs:288:14
    |
286 |           match self
    |  _______________-
287 | |             .state
288 | |             .compare_exchange(WAITING, REGISTERING, Acquire, Acquire)
    | |             -^^^^^^^^^^^^^^^^ method not found in `AtomicUsize`
    | |_____________|
    |

error[E0599]: no method named `compare_exchange` found for struct `AtomicUsize` in the current scope
   --> C:\Users\khsdfyd\.cargo\registry\src\rsproxy.cn-0dccff568467c15b\atomic-waker-1.1.2\src\lib.rs:313:26
    |
311 |                       let res = self
    |  _______________________________-
312 | |                         .state
313 | |                         .compare_exchange(REGISTERING, WAITING, AcqRel, Acquire);
    | |                         -^^^^^^^^^^^^^^^^ method not found in `AtomicUsize`
    | |_________________________|
    |

error[E0599]: no method named `swap` found for struct `AtomicUsize` in the current scope
   --> C:\Users\khsdfyd\.cargo\registry\src\rsproxy.cn-0dccff568467c15b\atomic-waker-1.1.2\src\lib.rs:338:40
    |
338 | ...                   self.state.swap(WAITING, AcqRel);
    |                                  ^^^^ method not found in `AtomicUsize`

error[E0599]: no method named `fetch_or` found for struct `AtomicUsize` in the current scope
   --> C:\Users\khsdfyd\.cargo\registry\src\rsproxy.cn-0dccff568467c15b\atomic-waker-1.1.2\src\lib.rs:403:26
    |
403 |         match self.state.fetch_or(WAKING, AcqRel) {
    |                          ^^^^^^^^ method not found in `AtomicUsize`

error[E0599]: no method named `fetch_and` found for struct `AtomicUsize` in the current scope
   --> C:\Users\khsdfyd\.cargo\registry\src\rsproxy.cn-0dccff568467c15b\atomic-waker-1.1.2\src\lib.rs:409:28
    |
409 |                 self.state.fetch_and(!WAKING, Release);
    |                            ^^^^^^^^^ method not found in `AtomicUsize`

   Compiling enumset_derive v0.8.1
For more information about this error, try `rustc --explain E0599`.
error: could not compile `atomic-waker` (lib) due to 5 previous errors
warning: build failed, waiting for other jobs to finish...
```

bjoernQ, 2023-12-20:
Most probably your problems are caused by the fact that you are using esp-wifi 0.1.1 from crates.io and HAL 0.14 from git
Because there is a tight coupling between them you need to use matching versions of esp-wifi and esp-hal

In this case it's a bit of a problem since we haven't release esp-wifi 0.2.0 yet because some of our dependencies need to get published first ( https://github.com/esp-rs/esp-wifi/issues/388 ) 

You either need to use the current `main` of esp-wifi or use esp32c3-hal version 0.13.0

khsdfyd, 2023-12-21:
yes，I modified `src/Cargo.toml` and `.cargo/config.toml`, and updated the versions of `esp-wifi` and `hal`, but new issues have arisen. Please provide more specific information about the new problems, such as error messages or descriptions of abnormal behavior, so that I can better assist you in resolving them.
`src/main.rs`
```
[target.riscv32imc-unknown-none-elf]
runner = "espflash flash --monitor"
linker = "rust-lld.exe"
[env]
ESP_LOGLEVEL="INFO"
[build]
rustflags = [
  "-C", "link-arg=-Tlinkall.x",
  "-C", "link-arg=-Trom_functions.x",
  "-C", "force-frame-pointers",
  "-C", "target-feature=+a",
  "--cfg", "target_has_atomic_load_store",
  "--cfg", 'target_has_atomic_load_store="8"',
  "--cfg", 'target_has_atomic_load_store="16"',
  "--cfg", 'target_has_atomic_load_store="32"',
  "--cfg", 'target_has_atomic_load_store="ptr"',
  "--cfg", "target_has_atomic",
  "--cfg", 'target_has_atomic="8"',
  "--cfg", 'target_has_atomic="16"',
  "--cfg", 'target_has_atomic="32"',
  "--cfg", 'target_has_atomic="ptr"',
]
target = "riscv32imc-unknown-none-elf"
[unstable]
build-std = ["alloc", "core"]
```
`src/Cargo.toml`
```
[dependencies]
embassy-executor = { version = "0.3.0", features = ["nightly", "integrated-timers", "arch-riscv32", "executor-thread"] }
embassy-time   = { version = "0.1.3", optional = true , features = ["tick-hz-16_000_000", "nightly"]}
hal = { package = "esp32c3-hal", version = "0.13.0", features = ["embassy", "embassy-time-timg0", "async", "rt"]}
esp-backtrace = { version = "0.9.0", features = ["esp32c3", "panic-handler", "exception-handler", "print-uart"] }
esp-println = { version = "0.7.0", features = ["esp32c3","log"] }
log = { version = "0.4.18" }
esp-alloc = { version = "0.3.0" }
esp-wifi  = { version = "0.1.1", features = ["esp32c3", "wifi", "ble", "esp-now"] }
smoltcp = { version = "0.10.0", default-features=false, features = ["proto-igmp", "proto-ipv4", "socket-tcp", "socket-icmp", "socket-udp", "medium-ethernet", "proto-dhcpv4", "socket-raw", "socket-dhcpv4"] }
embedded-svc = { version = "0.25.0", default-features = false, features = [] }
embedded-io = "0.4.0"
heapless = { version = "0.7.14", default-features = false }
critical-section   = "1.1.2"
embassy-sync       = "0.4.0"
static_cell        = { version = "2.0.0", features = ["nightly"] }
embedded-io-async = { version = "0.6.1", features = ["alloc"] }
```
this is output
```
   Compiling core v0.0.0 (C:\Users\khsdfyd\.rustup\toolchains\nightly-x86_64-pc-windows-msvc\lib\rustlib\src\rust\library\core)
   Compiling rustc-std-workspace-core v1.99.0 (C:\Users\khsdfyd\.rustup\toolchains\nightly-x86_64-pc-windows-msvc\lib\rustlib\src\rust\library\rustc-std-workspace-core)
   Compiling compiler_builtins v0.1.103
   Compiling alloc v0.0.0 (C:\Users\khsdfyd\.rustup\toolchains\nightly-x86_64-pc-windows-msvc\lib\rustlib\src\rust\library\alloc)
   Compiling critical-section v1.1.2
   Compiling byteorder v1.5.0
   Compiling stable_deref_trait v1.2.0
   Compiling nb v1.1.0
   Compiling futures-core v0.3.29
   Compiling cfg-if v1.0.0
   Compiling futures-task v0.3.29
   Compiling pin-utils v0.1.0
   Compiling void v1.0.2
   Compiling pin-project-lite v0.2.13
   Compiling embedded-hal v1.0.0-rc.1
   Compiling embedded-io v0.6.1
   Compiling bit_field v0.10.2
   Compiling log v0.4.20
   Compiling gcd v2.3.0
[... 133 lines cut ...]
          >>> referenced by embassy_executor.a5f14002e5b1c4cb-cgu.0
          >>>               embassy_executor-8c4d1ef5c87f7974.embassy_executor.a5f14002e5b1c4cb-cgu.0.rcgu.o:(embassy_executor::raw::waker::wake::h5e8f752086472ceb) in archive D:\Project\Rust\reyids\target\riscv32imc-unknown-none-elf\release\deps\libembassy_executor-8c4d1ef5c87f7974.rlib
          >>> referenced 5 more times

          rust-lld: error: undefined symbol: __sync_fetch_and_and_4
          >>> referenced by embassy_executor.a5f14002e5b1c4cb-cgu.0
          >>>               embassy_executor-8c4d1ef5c87f7974.embassy_executor.a5f14002e5b1c4cb-cgu.0.rcgu.o:(embassy_executor::raw::Executor::poll::h2eb74bc93813d107) in archive D:\Project\Rust\reyids\target\riscv32imc-unknown-none-elf\release\deps\libembassy_executor-8c4d1ef5c87f7974.rlib
          >>> referenced by embassy_executor.a5f14002e5b1c4cb-cgu.0
          >>>               embassy_executor-8c4d1ef5c87f7974.embassy_executor.a5f14002e5b1c4cb-cgu.0.rcgu.o:(embassy_executor::raw::Executor::poll::h2eb74bc93813d107) in archive D:\Project\Rust\reyids\target\riscv32imc-unknown-none-elf\release\deps\libembassy_executor-8c4d1ef5c87f7974.rlib
          >>> referenced by embassy_executor.a5f14002e5b1c4cb-cgu.0
          >>>               embassy_executor-8c4d1ef5c87f7974.embassy_executor.a5f14002e5b1c4cb-cgu.0.rcgu.o:(embassy_executor::raw::Executor::poll::h2eb74bc93813d107) in archive D:\Project\Rust\reyids\target\riscv32imc-unknown-none-elf\release\deps\libembassy_executor-8c4d1ef5c87f7974.rlib
          >>> referenced 1 more times

          rust-lld: error: undefined symbol: __sync_lock_test_and_set_4
          >>> referenced by embassy_executor.a5f14002e5b1c4cb-cgu.0
          >>>               embassy_executor-8c4d1ef5c87f7974.embassy_executor.a5f14002e5b1c4cb-cgu.0.rcgu.o:(embassy_executor::raw::Executor::poll::h2eb74bc93813d107) in archive D:\Project\Rust\reyids\target\riscv32imc-unknown-none-elf\release\deps\libembassy_executor-8c4d1ef5c87f7974.rlib

          rust-lld: error: undefined symbol: __sync_fetch_and_or_4
          >>> referenced by embassy_executor.a5f14002e5b1c4cb-cgu.0
[... cut ...] ct\Rust\reyids\target\riscv32imc-unknown-none-elf\release\deps\libembassy_executor-8c4d1ef5c87f7974.rlib
          >>> referenced by reyids.7010bf9253459859-cgu.0
          >>>               D:\Project\Rust\reyids\target\riscv32imc-unknown-none-elf\release\deps\reyids-aad288976adc10f1.reyids.7010bf9253459859-cgu.0.rcgu.o:(embassy_executor::raw::TaskStorage$LT$F$GT$::poll::h764ee98ba7227e52 (.llvm.17310760701440962987))

error: could not compile `reyids` (bin "reyids") due to 1 previous error
```

bjoernQ, 2023-12-21:
There are still a few dependencies which don't match

Have a look here: https://github.com/bjoernQ/esp32c3-ble-hid/blob/main/Cargo.toml

e.g. `static_cell` need to be pinned to "=1.2.0"

Also make sure to do a `cargo clean`,`cargo update` after changing the dependencies

khsdfyd, 2023-12-22:
Thanks to your invaluable assistance and guidance, I have been able to successfully resolve the issues I was facing. 
This my config
`src/.cargo/config.toml`
```
[target.riscv32imc-unknown-none-elf]
runner = "espflash flash --monitor"
[env]
ESP_LOGLEVEL="INFO"
[build]
rustflags = [
  "-C", "link-arg=-Tlinkall.x",
  "-C", "link-arg=-Trom_functions.x",
  "-C", "force-frame-pointers",
  "-C", "target-feature=+a",
  "--cfg", "target_has_atomic_load_store",
  "--cfg", 'target_has_atomic_load_store="8"',
  "--cfg", 'target_has_atomic_load_store="16"',
  "--cfg", 'target_has_atomic_load_store="32"',
  "--cfg", 'target_has_atomic_load_store="ptr"',
  "--cfg", "target_has_atomic",
  "--cfg", 'target_has_atomic="8"',
  "--cfg", 'target_has_atomic="16"',
  "--cfg", 'target_has_atomic="32"',
  "--cfg", 'target_has_atomic="ptr"',
]
target = "riscv32imc-unknown-none-elf"
[unstable]
build-std = ["alloc", "core"]
```
`src/Cargo.toml`
```
[dependencies]
embassy-executor = { version = "=0.3.2", features = [
    "nightly",
    "executor-thread",
    "integrated-timers",
    "arch-riscv32",
] }
embassy-time   = { version = "0.1.3", optional = true , features = ["tick-hz-16_000_000", "nightly"]}
hal = { package = "esp32c3-hal", version = "0.13.0", features = ["embassy", "embassy-time-timg0", "async", "rt"]}
esp-backtrace = { version = "0.9.0", features = ["esp32c3", "panic-handler", "exception-handler", "print-uart"] }
esp-println = { version = "0.7.0", features = ["esp32c3","log"] }
log = { version = "0.4.18" }
esp-alloc = { version = "0.3.0" }
esp-wifi  = { version = "0.1.1", features = ["esp32c3", "wifi", "ble", "esp-now"] }
smoltcp = { version = "0.10.0", default-features=false, features = ["proto-igmp", "proto-ipv4", "socket-tcp", "socket-icmp", "socket-udp", "medium-ethernet", "proto-dhcpv4", "socket-raw", "socket-dhcpv4"] }
embedded-svc = { version = "0.25.0", default-features = false, features = [] }
embedded-io = "0.4.0"
heapless = { version = "0.7.14", default-features = false }
critical-section   = "1.1.2"
embassy-sync       = "0.4.0"
static_cell = { version = "=1.2.0", features = ["nightly"] }
embedded-io-async = { version = "0.6.1", features = ["alloc"] }
[profile.dev.package.esp-wifi]
opt-level = 3
```

closed 2023-12-22 by khsdfyd

katyo, 2024-10-19:
I still have same issue with stable Rust.

```
error[E0599]: no method named `compare_exchange` found for struct `AtomicUsize` in the current scope
   --> /home/kayo/.cargo/registry/src/index.crates.io-6f17d22bba15001f/atomic-waker-1.1.2/src/lib.rs:288:14
    |
286 |           match self
    |  _______________-
287 | |             .state
288 | |             .compare_exchange(WAITING, REGISTERING, Acquire, Acquire)
    | |             -^^^^^^^^^^^^^^^^ method not found in `AtomicUsize`
    | |_____________|
    |

error[E0599]: no method named `compare_exchange` found for struct `AtomicUsize` in the current scope
   --> /home/kayo/.cargo/registry/src/index.crates.io-6f17d22bba15001f/atomic-waker-1.1.2/src/lib.rs:313:26
    |
311 |                       let res = self
    |  _______________________________-
312 | |                         .state
313 | |                         .compare_exchange(REGISTERING, WAITING, AcqRel, Acquire);
    | |                         -^^^^^^^^^^^^^^^^ method not found in `AtomicUsize`
    | |_________________________|
    |

error[E0599]: no method named `swap` found for struct `AtomicUsize` in the current scope
   --> /home/kayo/.cargo/registry/src/index.crates.io-6f17d22bba15001f/atomic-waker-1.1.2/src/lib.rs:338:40
    |
338 | ...                   self.state.swap(WAITING, AcqRel);
    |                                  ^^^^ method not found in `AtomicUsize`

error[E0599]: no method named `fetch_or` found for struct `AtomicUsize` in the current scope
   --> /home/kayo/.cargo/registry/src/index.crates.io-6f17d22bba15001f/atomic-waker-1.1.2/src/lib.rs:403:26
    |
403 |         match self.state.fetch_or(WAKING, AcqRel) {
    |                          ^^^^^^^^ method not found in `AtomicUsize`

error[E0599]: no method named `fetch_and` found for struct `AtomicUsize` in the current scope
   --> /home/kayo/.cargo/registry/src/index.crates.io-6f17d22bba15001f/atomic-waker-1.1.2/src/lib.rs:409:28
    |
409 |                 self.state.fetch_and(!WAKING, Release);
    |                            ^^^^^^^^^ method not found in `AtomicUsize`

For more information about this error, try `rustc --explain E0599`.
error: could not compile `atomic-waker` (lib) due to 5 previous errors
```

```
$ rustc --version
rustc 1.82.0 (f6e511eec 2024-10-15)
```