esp-rs/esp-hal #1052: How to build a custom bootloader for a 26mhz esp32c2?

AlexeyTuboltsev, 2023-12-26:
I'm trying to build an example from esp-wifi for my esp32c2 with a 26mhz crystal. After running through some hoops to pass the "xtal-26mhz" feature from esp-wifi's Cargo.toml down to esp-hal/esp32c3-hal I finally managed to build the code but now I get a boot loop with this error:
```
  !! A panic occured in '/home/lexey/esp-hal/esp-hal-common/src/rtc_cntl/rtc/esp32c2.rs', at line 65, column 5
                                                                                                               
PanicInfo {
    payload: Any { .. },
    message: Some( Did you flash the right bootloader configured for 26MHz xtal?, ),
    location: Location {
          file: "/home/lexey/esp-hal/esp-hal-common/src/rtc_cntl/rtc/esp32c2.rs", line: 65,  col: 5, },
          can_unwind: true,
          force_no_backtrace: false,
      }
    
    Backtrace:  0x4201e65c
                      0x420000d0
```

In the original pull request where this feature was added author mentions that a custom bootloader is necessary (https://github.com/esp-rs/esp-hal/pull/301). How do I make one?
Thank you!

bjoernQ, 2023-12-27:
Easiest way to build a suitable bootloader is to use esp-idf and build one of the examples (which one doesn't matter -  `examples/get-started/hello_world` is probably a good fit)

Do `esptool.py set-target esp32c2`

Before building you need to use `menuconfig` to set `Component config → Hardware Settings → Main XTAL Config → Main XTAL frequency` to 26MHz (40 by default)

After `idf.py build` you should find the bootloader as `examples/get-started/hello_world/build/bootloader/boot
loader.bin`

That can be used with `espflash`

AlexeyTuboltsev, 2023-12-27:
Thank you very much! I'll give it a try later today

SergioGasquez, 2023-12-29:
See https://esp-rs.github.io/book/troubleshooting/espflash.html#rtc_clk_init-possibly-invalid-config_xtal_freq-setting and https://github.com/esp-rs/espflash/issues/526

closed 2023-12-29 by SergioGasquez