esp-rs/esp-hal #1056: esp32c3: example/embassy_wait: GPIO 9 doesn't work, but GPIO 1 and 20 do

harryhaaren, 2024-01-01:
Running this example does not work (no `Ping!` prints show up).
https://github.com/esp-rs/esp-hal/blob/2e2d67f315d7c3abf0d023fda541473858769965/esp32c3-hal/examples/embassy_wait.rs#L36

Changing the `gpio9` to `gpio20` (the pin just beside it) works, and using `gpio1` works fine too, on the `ESP32-C3-DevKit-RUST-1`.
This might be a board malfunction, but perhaps somebody could check it is working there? (A [quick search](https://github.com/esp-rs/esp-hal/issues?q=is%3Aissue+is%3Aopen+c3+wait) didn't show up anything here..)

bugadani, 2024-01-01:
This might be a duplicate of https://github.com/esp-rs/esp-hal/issues/657 but the info that certain pins work while others don't is interesting for sure.

bjoernQ, 2024-01-02:
Did you connect an additional button to GPIO9? GPIO9 is the boot button and we use it to allow testing the example without additional wires/buttons

Does the `gpio_interrupt.rs` example work for you (with GPIO9)?

jessebraham, 2024-01-02:
On the ESP Rust Board, GPIO9 has an external 10k pull-up resistor connected. So you have an external pull-up and an internal pull-down on the same GPIO. I imagine this probably has something to do with your issues.

harryhaaren, 2024-01-02:
> This might be a duplicate of #657 but the info that certain pins work while others don't is interesting for sure.

I read that issue, and do not believe it to be the same root-cause.

> Did you connect an additional button to GPIO9? GPIO9 is the boot button and we use it to allow testing the example without additional wires/buttons

Yes, I was using some wires to act as a switch/button. I wasn't aware the `boot` button on this device is *also* on `gpio9`: learning moment for me.  Re-testing with the `boot` button it works perfectly - and no external HW required; nice one.

Closing - root cause is "user error" in that I wasn't aware `gpio9` was actually the boot button. That said, the docs don't call it out, and the "external button" approach doesn't work (likely the potential problem @jessebraham [described above](https://github.com/esp-rs/esp-hal/issues/1056#issuecomment-1874173524))

Thanks for the replies folks!

closed 2024-01-02 by harryhaaren