esp-rs/esp-hal #1077: [ESP32] Inverted ADC Values

hansingt, 2024-01-12:
Hi, I receive unexpected ADC readings from my ESP32-Wrover-E (ESP32-D0WDQ5 revision 3.1) when using the following code:
```rust
#![no_std]
#![no_main]

use esp_backtrace as _;
use esp_println::println;
use hal::{
    adc::{Attenuation, ADC, AdcConfig},
    analog::AnalogExt,
    clock::ClockControl,
    peripherals::Peripherals,
    prelude::*,
    system::SystemExt,
    Delay, IO,
};

#[entry]
fn main() -> ! {
    let peripherals = Peripherals::take();
    let system = peripherals.SYSTEM.split();
    let io = IO::new(peripherals.GPIO, peripherals.IO_MUX);
    let analog = peripherals.SENS.split();

    // Configure the system clock
    let clocks = ClockControl::max(system.clock_control).freeze();

    // Configure the ADC
    let mut adc_config = AdcConfig::default();
    let mut adc_pin =
        adc_config.enable_pin(io.pins.gpio25.into_analog(), Attenuation::Attenuation11dB);
    let mut adc = ADC::adc(analog.adc2, adc_config.into()).unwrap();

    let mut delay = Delay::new(&clocks);
    loop {
        let sample = nb::block!(adc.read(&mut adc_pin)).unwrap();
        println!("ADC Value: {sample}");
        delay.delay_ms(200u32);
    }
}
```
For me, this results in inverted values for the ADC, e.g. 0V being 4095 and 3.3V being 0. Is this expected? I couldn't find any notice in the reference manual. But I found an issue about this in the official esp-idf repo: https://github.com/espressif/esp-idf/issues/4557 this says, that the data inversion bit should be set and that it will be added to the IDF framework on next release.

Doing this, fixes the issue for me. Maybe, this is missing in ADC initialization code for the ESP32?
```rust
    peripherals
        .SENS
        .sar_read_ctrl2()
        .modify(|_, w| w.sar2_data_inv().set_bit());
```

Btw: This happens on both ADCs and on every channel.

jessebraham, 2024-01-17:
Thanks for the issue. I'm currently working on some refactoring of the `analog` module anyway, so I will try to get this fixed while I'm at it.

closed 2024-01-19 by bjoernQ

MauroMombelli, 2025-04-13:
this seems to be incorrect for let analog_pin = peripherals.GPIO34;
using a devkit v4

PR #1093 (2024-01-19): Do not invert ADC readings when using ESP32
Fixes #1077

This feels backwards to me (I would expect that the bit needs clearing, instead) however I did confirm that this fixes the issue on hardware. @MabezDev @bjoernQ would one of you mind just verifying this, please?