esp-rs/esp-hal #1083: Static linking in psram. Is there support or what is needed for it?

Volkalex28, 2024-01-16:
Hello. Are there any plans to add a linking attribute to psram? 
As in idf:
`#[link_section = ".ext_ram.bss.****"]`

bjoernQ, 2024-01-16:
There are currently no plans to use PSRAM for anything but heap memory. PRs are welcome

Volkalex28, 2024-01-16:
Why was it removed [here](https://github.com/esp-rs/esp-hal/commit/92bda00296c0b31d77c39ac095330a8d464d2bc3)?

bjoernQ, 2024-01-16:
> Why was it removed [here](https://github.com/esp-rs/esp-hal/commit/92bda00296c0b31d77c39ac095330a8d464d2bc3)?

It was never really supported ... e.g. we never actually zeroed `.external.bss`

Volkalex28, 2024-01-17:
Added a [draft ](https://github.com/esp-rs/esp-hal/pull/1090) for this. I will be glad for any help

ProfFan, 2024-08-24:
I had some discussion with my coworkers today and realized that there is a huge benefit to this (PSRAM XIP): The PSRAM has much lower latency than the flash, so ICACHE misses will be much faster! XREF #1096 for example. Also per official docs: https://docs.espressif.com/projects/esp-idf/en/stable/esp32s3/api-reference/kconfig.html#config-spiram-fetch-instructions

Tagging @MabezDev since I saw the comments in #1090

igrr, 2024-09-04:
> The PSRAM has much lower latency than the flash, so ICACHE misses will be much faster!

@ProfFan I am curious about this conclusion, could you please explain it further? AFAICT, provided that Flash and PSRAM use same SPI bus width and frequency, read latencies would be the same (plus-minus a couple of dummy cycles after the address phase).

ProfFan, 2024-09-04:
@igrr 
> I am curious about this conclusion

It's written in the official docs: https://docs.espressif.com/projects/esp-idf/en/stable/esp32s3/api-reference/kconfig.html#config-spiram-fetch-instructions

Also quote from docs:

> The benefits of XiP from PSRAM is:

> PSRAM access speed is faster than Flash access. So the performance is better.
The cache will not be disabled during an SPI1 flash operation, thus optimizing the code execution performance during SPI1 flash operations. For ISRs, ISR callbacks and data which might be accessed during this period, you do not need to place them in internal RAM, thus internal RAM usage can be optimized. This feature is useful for high throughput peripheral involved applications to improve the performance during SPI1 flash operations.

Be-ing, 2025-04-05:
It doesn't seem the non-global allocator API will be stabilized upstream anytime soon, so currently putting anything in PSRAM on a stable compiler requires esp-alloc with the global allocator using PSRAM. If you also want a heap for internal RAM, you can't with a stable compiler. If specific statics could be put into PSRAM, that could sidestep this dilemma.

MarSik, 2026-09-24:
The allocator API just stabilized in https://github.com/rust-lang/rust/issues/32838 / https://github.com/rust-lang/rust/pull/156882