esp-rs/esp-hal #1087: Use the "generic-queue" from embassy-time instead of the "integrated-timers" from embassy-executor

Fundevoge, 2024-01-17:
https://github.com/esp-rs/esp-hal/discussions/1037

The "integrated-timers" timer queue from embassy-executor does not work properly on esp32.
I added "generic-queue" from embassy-time and got a linker error.
As it turns out, this has to do with the  "embassy-integrated-timers" default-feature of the esp32-hal.

With this default feature, embassy-executor creates the _embassy_time_schedule_wake function:
```rs
#[cfg(feature = "integrated-timers")]
embassy_time_queue_driver::timer_queue_impl!(static TIMER_QUEUE: TimerQueue = TimerQueue);
```
embassy-time provides an alternative timer_queue_impl behind "generic-queue":
```rs
#[cfg(feature = "generic-queue")]
embassy_time_queue_driver::timer_queue_impl!(static QUEUE: Queue = Queue::new());
```

But if both features are enabled, this duplicate declaration results in a nasty link-time error...
```rs
extern "Rust" {
    fn _embassy_time_schedule_wake(at: u64, waker: &Waker);
}

#[macro_export]
macro_rules! timer_queue_impl {
    (static $name:ident: $t: ty = $val:expr) => {
        static $name: $t = $val;

        #[no_mangle]
        fn _embassy_time_schedule_wake(at: u64, waker: &core::task::Waker) {
            <$t as $crate::TimerQueue>::schedule_wake(&$name, at, waker);
        }
    };
}
```

My proposed solution is to use the "generic-queue" from embassy-time instead of the "integrated-timers" from embassy-executor or to at least remove "embassy-integrated-timers" from the default features.

bjoernQ, 2024-01-17:
Running into linker errors is not nice.

We probably should improve the documentation regarding the features (i.e. explain that `embassy-integrated-timers` is a default feature and cannot be used together with `embassy-generic-queue-*`) and have a check in `build.rs` giving a nicer error if those features are activated concurrently

Fundevoge, 2024-01-17:
That sounds like a good idea. I am also a bit confused whether to add i.e. `embassy-generic-queue-8` to esp32-hal or to esp-hal-common or to embassy-time as `generic-queue-8`, as they all seem to do about the same. Maybe also add documentation on that. If the feature is enabled in embassy-time directly, I am not sure if the `build.rs` of esp32-hal / esp-hal-common could see that and give an error.

bjoernQ, 2024-01-17:
Good point! Given with #1089 we'll probably remove the `generic-queue-*` features and let the user enable them on `embassy-time` we probably need to rethink this

At least we have an issue reminding us that we should do something about the ergonomics here 👍

MabezDev, 2024-02-28:
As part of #1196 we no longer enable integrated timers by default, leaving the end user with the choice of using generic queue or integrated timers.