esp-rs/esp-hal #1110: #[ram(uninitialized)] statics are unsound

icewind1991, 2024-01-23:
```rust
#[entry]
fn main() -> ! {
    ...
    print(&BOOL);
    ...
}

// usage of the static is done trough a function call to stop rustc from just inlining the constant
// release mode will inline things enough that it just prints the constant
// more complex code might still fail in release mode
fn print(val: &bool) {
    println!("{:?}", *val as u32);
}

#[ram(rtc_fast, uninitialized)]
static BOOL: bool = false;
```

will output a number that isn't a 0 or 1 which means that any use of the boolean is undefined behavior.

While in practice this is probably *fine* since the main use case for `#[ram(uninitialized)]` is through `static mut` where access is `unsafe` anyway this can still lead to unexpected issues when checking the `static` before writing it or when dealing with atomics.

Imo the best approach would be to have the macro require that the type of the statics are 
["plain old data"](https://docs.rs/bytemuck/latest/bytemuck/trait.Pod.html#safety) to ensure that all bit patterns are valid for the type.

wetheredge, 2024-04-22:
> Imo the best approach would be to have the macro require that the type of the statics are
["plain old data"](https://docs.rs/bytemuck/latest/bytemuck/trait.Pod.html#safety) to ensure that all bit patterns are valid for the type.

My reading of the following excerpt from [the docs for `MaybeUninit`](https://doc.rust-lang.org/std/mem/union.MaybeUninit.html#initialization-invariant) is that even that would be unsound:

> Uninitialized memory is special in that it does not have a fixed value (“fixed” meaning “it won’t change without being written to”). Reading the same uninitialized byte multiple times can give different results. This makes it undefined behavior to have uninitialized data in a variable even if that variable has an integer type, which otherwise can hold any fixed bit pattern:
>
> ```rust
> use std::mem::{self, MaybeUninit};
>
> let x: i32 = unsafe { mem::uninitialized() }; // undefined behavior! ⚠️
> // The equivalent code with `MaybeUninit<i32>`:
> let x: i32 = unsafe { MaybeUninit::uninit().assume_init() }; // undefined behavior! ⚠️
> ```

My approach is to use `MaybeUninit<T>` and check [`esp_hal::reset::get_reset_reason`](https://docs.esp-rs.org/esp-hal/esp-hal/0.17.0/esp32s3/esp_hal/reset/fn.get_reset_reason.html), reinitializing the static for anything but `CoreSw`. As long as nothing calls [`esp_hal::reset::software_reset()`](https://docs.esp-rs.org/esp-hal/esp-hal/0.17.0/esp32s3/esp_hal/reset/fn.software_reset.html) before that check runs, it should prevent ever getting access to uninitialized memory.

I can post my code if that explanation of it doesn't make sense.

bjoernQ, 2024-04-22:
Maybe the HAL could check the reset reset-reason during start-up and initialize that memory region if needed (and not touching it otherwise) 🤔

wetheredge, 2024-04-24:
Seems like that should work. I can think of 3 options:

1. Have the `#[ram(uninitialized)]` macro wrap the static in a `MaybeUninit` with a safety comment documenting what had to be done to use it soundly.
2. Wrap it in a new type that has a method that handles the check & initialize. It would still have to be `unsafe` with documentation saying to always call it before any possible resets and to handle synchronization, etc., but it would be a little harder to mess up.
3. As you suggested: Automatically insert the check & initialization before/at the beginning of `#[entry]` somehow. Are there any interrupts that could be set up like [the direct vectoring example](https://github.com/esp-rs/esp-hal/blob/e0610f799722fe5286985456d581e5f0610f0cbc/examples/src/bin/direct_vectoring.rs) that don't need to be enabled by the user and could potentially run before the check?

If any of those sound reasonable I could start working on an implementation this week.

bjoernQ, 2024-04-24:
I'd say option 3 probably results in the best developer-experience.

Some interrupts are enabled by the second stage bootloader but very early we disable all known interrupts - if any interrupt triggers before user-main that would be a bug. (in that case we couldn't even guarantee that anything is initialized at all)

I'm just wondering if `uninitialized` is the correct term, then? 

Maybe keeping `uninitialized` as is with a scary warning in the docs and adding a new option (don't have an idea how to name that) which initializes the memory only when needed would be an idea?

Thanks for volunteering working on this!

Any thoughts @MabezDev ?

wetheredge, 2024-04-25:
Maybe `persistent`?

I was thinking of an option to allow reducing the list of reset reasons that skip initialization. Perhaps `#[ram(…, persistent)]` to use the default list and `#[ram(…, persistent(CoreSw)]` or `#[ram(…, persistent(after = CoreSw)]` to only skip initialization for `CoreSw`?

I'd vote for removing `uninitialized` completely, since any sound use of it would have to reinvent everything in this issue. If anyone *really* wants to implement it themselves, they could use `#[link_section = ".rtc_(fast|slow).noinit"]`.

bjoernQ, 2024-04-25:
> Maybe `persistent`?
> 
> I was thinking of an option to allow reducing the list of reset reasons that skip initialization. Perhaps `#[ram(…, persistent)]` to use the default list and `#[ram(…, persistent(CoreSw)]` or `#[ram(…, persistent(after = CoreSw)]` to only skip initialization for `CoreSw`?
> 

At least I have no better idea 👍 

> I'd vote for removing `uninitialized` completely, since any sound use of it would have to reinvent everything in this issue. If anyone _really_ wants to implement it themselves, they could use `#[link_section = ".rtc_(fast|slow).noinit"]`.

True ... I can think of uses for truly uninitialized memory in general but it's probably rare and it's possible to specify the link section via `link_section`

PR #1677 (2024-07-10): Take 2 on `#[ram]` soundness
### Submission Checklist 📝
- [x] I have updated existing examples or added new ones (if applicable).
- [x] I have used `cargo xtask fmt-packages` command to ensure that all changed code is formatted correctly.
- [ ] My changes were added to the [`CHANGELOG.md`](https://github.com/esp-rs/esp-hal/blob/main/esp-hal/CHANGELOG.md) in the **_proper_** section.
- [x] My changes are in accordance to the [esp-rs API guidelines](https://github.com/esp-rs/esp-hal/blob/main/API-GUIDELINES.md)

#### Extra:
- [x] I have read the [CONTRIBUTING.md guide](https://github.com/esp-rs/ [... cut]
closed 2024-07-10 by jessebraham