esp-rs/esp-hal #1117: Consider using a changelog directory instead of editing CHANGELOG.md directly?

bjoernQ, 2024-01-26:
Editing CHANGELOG.md creates a lot of merge conflicts - maybe we should try doing what probe-rs is doing with https://github.com/probe-rs/probe-rs/tree/master/changelog

Seems they have an xtask for checking the changelog in PRs and assembling the changelog: https://github.com/probe-rs/probe-rs/blob/master/xtask/src/main.rs

jessebraham, 2024-01-26:
I don't really see the need personally, I've never really found updating `CHANGELOG.md` to be that problematic personally. I always do a cleanup pre-release, too.

If this is something that actually bothers people I can somebody can set this up, I found it confusing contributing to `probe-rs` the first time though so we will need some docs (that nobody will read 😁).

bjoernQ, 2024-01-26:
Generally, it's also not an actual problem for me but there are two cases where it needs some care (not saying solving it is worth it ... I agree that the `changelog` dir is somewhat confusing)

- when there was a release after opening the PR it's easy to forget to move the entry
- sometimes it's "hard" to remember to add to the unreleased section (t.b.h. should be a no-brainer but it happened to at least three people including me)

I'm really not too much in favor of the changelog-directory myself - just wanted to start a discussion

bugadani, 2024-01-29:
probe-rs's solution has one big quality of life element: it's not necessary to manually specify the PR number. This makes it at least one step more convenient - after people learn how to use it. After getting used to, well I may not prefer it, but I find it to be a pretty good solution.

Thankfully people only need to bump against the idea once and they then seem to remember. Maybe because it's different? :)

Dominaezzz, 2024-01-29:
I'd be happy with leaving the CHANGELOG update to the last commit, as that prevents conflicts while iterating on the PR. Unfortunately the big red X on my commits bother me too much to do that. :sweat_smile:.

bjoernQ, 2024-01-30:
> probe-rs's solution has one big quality of life element: it's not necessary to manually specify the PR number

Oh yes - that is quite convenient. Haven't thought about that

> Unfortunately the big red X on my commits bother me too much to do that. 

One thing we might should do at least is to not require it for draft PRs - I think we currently always check it

jessebraham, 2024-01-30:
I guess if somebody wants to set up whatever infra we need for this, then add a `CODEOWNERS` file please so that you can maintain it. Also please document the process somewhere, probably the top-level `README.md`.

I honestly don't really want to do this, but if it's better for other people than I guess it's fine and I'll just suck it up and deal with it.

MabezDev, 2024-02-19:
I think that a process change right now is probably not worth the effort. We can revisit this if we ever get too annoyed by the current solution.

closed 2024-02-19 by MabezDev