esp-rs/esp-hal #1123: The `twai` driver is tightly coupled to the `embedded-hal`/`embedded-can` packages

jessebraham, 2024-01-26:
Ideally, all of our drivers should function without using any of the `embedded-hal`/`embedded-can` traits; these traits are only interfaces. Unfortunately, our `twai` driver is currently quite tightly coupled to these dependencies.

This needs to be fixed; I have started looking into this a bit, but unfortunately I think we will need to duplicate quite a few enums/structs from these packages :/

jessebraham, 2024-01-26:
I don't really see a path forward here other than duplicating all the types from these packages and implementing the `From` trait as needed for everything. Seems this will be a bit painful.

ptpaterson, 2024-01-27:
I will say that using the provided structs and enums from `embedded_can` is rather convenient. I'm working on a generic CAN library, and it'd be great if I can use `ID` instead of `impl Into<ID>` everywhere. But I definitely agree peripheral traits shouldn't be required to use this HAL. #951 does create inherent impls for transmit and receive, which the trait impl then reuses. If the consensus is to remove dependence on `embedded_can` altogether, then that PR is an incremental step toward that.

jessebraham, 2024-03-22:
Let's use #1317 to track progress here moving forward.

closed 2024-03-22 by jessebraham