esp-rs/esp-hal #1127: esp32h2: default XTAL is 32mhz

MabezDev, 2024-01-30:
[image]

This doesn't seem to be causing problems right now, but the fact we don't have a feature for it, in esp-hal or espflash means we were probably not aware of this.

jessebraham, 2024-01-30:
As a side note, which has been raised before (more than once), it's incredibly unclear how to map from the clock diagrams in the TRMs to our clock code. Would be _really_ helpful having some documentation here, as clocks seem to be randomly selected from both the left and right edges of the diagram.

jessebraham, 2024-02-28:
Now that we've eliminated the XTAL frequency features I think we can close this? Unless I'm missing something the H2 seems to be using this frequency now.

@MabezDev please re-open if I am mistaken.

MabezDev, 2024-02-28:
The 32MHZ xtal case isnot yet handled so we need to keep this open.

jessebraham, 2024-02-28:
Can you elaborate please? What actually needs to be done? The XTAL frequency in boot defaults is 32MHz, where else does this need to be changed?

MabezDev, 2024-02-29:
I spent most of today trying to remember why I wanted to not forget about this, it was this issue I wanted to test: https://github.com/esp-rs/esp-hal/issues/766 and I think the xtal was the issue. We can close this now.

closed 2024-02-29 by MabezDev