esp-rs/esp-hal #1130: No implementation of `embedded_hal_async::spi::SpiDevice` in `esp32c3-hal`

reivilibre, 2024-01-30:
I can't see an implementation of `SpiDevice` in the `esp32c3-hal` crate.

I want one so that I can use `embassy-net-wiznet`:
https://github.com/embassy-rs/embassy/blob/dcce40c8a2eefb956ffadbfcc3db6c27cde55dab/embassy-net-wiznet/src/lib.rs#L42

I *can* see an implementation of `SpiBus`, but I'm actually confused as it seems like it might be more appropriate for it to be a `SpiDevice`?

That's because you pass in a CS (Chip Select) pin, at least in the `embassy_spi` example: https://github.com/esp-rs/esp-hal/blob/v0.15.0/esp32c3-hal/examples/embassy_spi.rs#L70

According to the docs for `SpiDevice`, '`SpiDevice` represents ownership over a single SPI device on a (possibly shared) bus, **selected with a CS (Chip Select) pin**',
whereas '`SpiBus` represents exclusive ownership **over the whole SPI bus**, with SCK, MOSI and MISO pins'.

I very well may has missed an implementation of `SpiDevice` somewhere so please do point it out if I'm wrong. (I looked in `cargo doc` and the list of Implementors is empty, though.)

bjoernQ, 2024-01-31:
According to https://docs.rs/embedded-hal/1.0.0/embedded_hal/spi/index.html#for-hal-authors

> HALs *must* implement [SpiBus](https://docs.rs/embedded-hal/1.0.0/embedded_hal/spi/trait.SpiBus.html). Users can combine the bus together with the CS pin (which should implement [OutputPin](https://docs.rs/embedded-hal/1.0.0/embedded_hal/digital/trait.OutputPin.html)) using HAL-independent [SpiDevice](https://docs.rs/embedded-hal/1.0.0/embedded_hal/spi/trait.SpiDevice.html) implementations such as the ones in [embedded-hal-bus](https://crates.io/crates/embedded-hal-bus).
>

> HALs may additionally implement [SpiDevice](https://docs.rs/embedded-hal/1.0.0/embedded_hal/spi/trait.SpiDevice.html) *to take advantage of hardware CS management*, which may provide some performance benefits. (There’s no point in a HAL implementing [SpiDevice](https://docs.rs/embedded-hal/1.0.0/embedded_hal/spi/trait.SpiDevice.html) if the CS management is software-only, this task is better left to the HAL-independent implementations).

I agree that we probably should implement SpiDevice since our hardware is capable of managing CS pins

TL;DR should be possible to use [embedded-hal-bus](https://crates.io/crates/embedded-hal-bus) with our implementation

reivilibre, 2024-01-31:
> TL;DR should be possible to use [embedded-hal-bus](https://crates.io/crates/embedded-hal-bus) with our implementation

This seems to be for `embedded-hal` rather than `embedded-hal-async`, unfortunately.

But you're right, that it seems like someone could make a very easy wrapper, particularly since `SpiBus` can manage the CS pin itself, the CS management could be a 'no-op'.

However it seems like this should be encoded directly in the type system of the ESP HAL: if the `SpiBus` has a CS pin attached, it ought to offer an `SpiDevice` interface (as well?).

bjoernQ, 2024-01-31:
> This seems to be for embedded-hal rather than embedded-hal-async, unfortunately.

There is an async feature: https://github.com/rust-embedded/embedded-hal/blob/04b4f1897a126d1f3aebf679489b60d8e994c839/embedded-hal-bus/Cargo.toml#L19

jessebraham, 2024-02-05:
I don't think there's really anything actionable here so going to go ahead and close the issue. If you disagree feel free to re-open it.

Notliam99, 2025-08-28:
# For All Stuck On This Issue:
I have found a Solution [https://esp32.implrust.com/spi/esp32-spi.html](https://esp32.implrust.com/spi/esp32-spi.html)