esp-rs/esp-hal #1136: Compilation Warning in esp32c3 (revision v0.4) Affecting Smartle , v0.3 Works Fine

Song-aff, 2024-02-03:
[2024-02-03T09:39:37Z INFO ] 🚀 A new version of espflash is available: v2.1.0
[2024-02-03T09:39:37Z INFO ] Serial port: 'COM6'
[2024-02-03T09:39:37Z INFO ] Connecting...
[2024-02-03T09:39:38Z INFO ] Using flash stub
Chip type:         esp32c3 (revision v0.4)
Crystal frequency: 40MHz
Flash size:        4MB
Features:          WiFi, BLE
MAC address:       ec:da:3b:d1:0a:78
App/part. size:    225,776/4,128,768 bytes, 5.47%
[00:00:00] [========================================]      13/13      0x0
[00:00:00] [========================================]       1/1       0x8000
[00:00:01] [========================================]      75/75      0x10000
[2024-02-03T09:39:40Z INFO ] Flashing has completed!
Commands:
    CTRL+R    Reset chip
    CTRL+C    Exit

ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x15 (USB_UART_CHIP_RESET),boot:0xc (SPI_FAST_FLASH_BOOT)
Saved PC:0x40380836
0x40380836 - core::option::Option<T>::unwrap
    at C:\Users\Administrator.DESKTOP-7JDKGIL\.rustup\toolchains\nightly-x86_64-pc-windows-msvc\lib\rustlib\src\rust\library\core\src\option.rs:929
SPIWP:0xee
mode:DIO, clock div:2
load:0x3fcd5820,len:0x171c
0x3fcd5820 - _stack_start
    at ??:??
load:0x403cc710,len:0x968
0x403cc710 -
    at ??:??
load:0x403ce710,len:0x2f68
0x403ce710 -
    at ??:??
SHA-256 comparison failed:
Calculated: 1d06b938c0222bf626e0bdf46178b1b37ab24d03f0360fc8fcf7153c2571deaf
Expected: 68d7bdf643ba446b8ed7ae8423241d442fd052b2bc77091100ba06fd65dcf8d5
Attempting to boot anyway...
entry 0x403cc710
0x403cc710 -
    at ??:??
I (24) boot: ESP-IDF v5.1-beta1-378-gea5e0ff298-dirt 2nd stage bootloader
I (24) boot: compile time Jun  7 2023 07:59:10
I (25) boot: chip revision: v0.4
I (29) boot.esp32c3: SPI Speed      : 40MHz
I (34) boot.esp32c3: SPI Mode       : DIO
I (38) boot.esp32c3: SPI Flash Size : 4MB
I (43) boot: Enabling RNG early entropy source...
I (49) boot: Partition Table:
I (52) boot: ## Label            Usage          Type ST Offset   Length
I (59) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (67) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (74) boot:  2 factory          factory app      00 00 00010000 003f0000
I (82) boot: End of partition table
I (86) esp_image: segment 0: paddr=00010020 vaddr=3c030020 size=065e4h ( 26084) map
I (100) esp_image: segment 1: paddr=0001660c vaddr=40380000 size=011fch (  4604) load
I (104) esp_image: segment 2: paddr=00017810 vaddr=00000000 size=08808h ( 34824)
I (119) esp_image: segment 3: paddr=00020020 vaddr=42000020 size=271a0h (160160) map
I (155) boot: Loaded app from partition at offset 0x10000
I (155) boot: Disabling RNG early entropy source...

MabezDev, 2024-02-03:
I don't see a warning here, could you elaborate?

MabezDev, 2024-02-19:
Closing, please re open if you find a reproducible error.

closed 2024-02-19 by MabezDev