esp-rs/esp-hal #1138: ESP32-S3 communication over usb stop working when wifi controller start

rpisarew, 2024-02-05:
Hello,

I have a issue with the wifi controller on a ESP32-S3. When the wifi controller starts the usb connection stops working and the tx led (for the usb) on the board is flashing rapidly. After that I am not able to connect with `espflash monitor` or flash the controller. The only way is to start with the boot button pressed, then I am able to flash or connect with the serial monitor again, until the wifi controller starts. Any idea why this is happening?

Dev Board: WROOM-1-N16R8 ESP32-S3-DevKitC-1
Rustc: v1.75.0-nightly
Repository to reproduce the issue: https://github.com/rpisarew/esp32-s3-wifi/tree/main

Output of the serial monitor:
```
I (45) boot: ESP-IDF v5.1-beta1-378-gea5e0ff298-dirt 2nd stage bootloader
I (46) boot: compile time Jun  7 2023 08:07:32
I (47) boot: Multicore bootloader
I (51) boot: chip revision: v0.2
I (55) boot.esp32s3: Boot SPI Speed : 40MHz
I (59) boot.esp32s3: SPI Mode       : DIO
I (64) boot.esp32s3: SPI Flash Size : 16MB
I (69) boot: Enabling RNG early entropy source...
I (74) boot: Partition Table:
I (78) boot: ## Label            Usage          Type ST Offset   Length
I (85) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (93) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (100) boot:  2 factory          factory app      00 00 00010000 00fa0000
I (108) boot: End of partition table
I (112) esp_image: segment 0: paddr=00010020 vaddr=3c0d0020 size=21174h (135540) map
I (154) esp_image: segment 1: paddr=0003119c vaddr=3fc8e020 size=0102ch (  4140) load
I (156) esp_image: segment 2: paddr=000321d0 vaddr=3fcc8168 size=00168h (   360) load
I (160) esp_image: segment 3: paddr=00032340 vaddr=40378000 size=0601ch ( 24604) load
I (176) esp_image: segment 4: paddr=00038364 vaddr=00000000 size=07cb4h ( 31924)
I (185) esp_image: segment 5: paddr=00040020 vaddr=42000020 size=c05d4h (787924) map
I (383) boot: Loaded app from partition at offset 0x10000
I (384) boot: Disabling RNG early entropy source...
embassy::init embassy-time-timg0
INFO - esp-wifi configuration Config { rx_queue_size: 5, tx_queue_size: 3, static_rx_buf_num: 10, dynamic_rx_buf_num: 32, static_tx_buf_num: 0, dynamic_tx_buf_num: 32, ampdu_rx_enable: 0, ampdu_tx_enable: 0, amsdu_tx_enable: 0, rx_ba_win: 6, max_burst_size: 1, country_code: "CN", country_code_operating_class: 0, mtu: 1492, heap_size: 65536, tick_rate_hz: 50, listen_interval: 3, beacon_timeout: 6, ap_beacon_timeout: 300, failure_retry_cnt: 1, scan_method: 0 }
DEBUG - wifi init
DEBUG - spawning task wifi: 0x4037a400 param 0x0 prio 253
DEBUG - task started: 0x4037a400 0x0
DEBUG - timer_disarm 3fcc7a78 not found
DEBUG - timer_setfn 3fcc7a04 0x420a83f4 0x0
DEBUG - timer_setfn 3fcc7a18 0x420a83e4 0x0
DEBUG - timer_setfn 3fcc7a54 0x420a83d4 0x0
DEBUG - timer_setfn 3fcc7a40 0x420a83c4 0x0
DEBUG - timer_setfn 3fcc7a78 0x420a83b4 0x0
DEBUG - timer_setfn 3fcc7a8c 0x420a83a4 0x0
DEBUG - timer_setfn 3fcc7aa0 0x420a8394 0x0
DEBUG - timer_setfn 3fcc7abc 0x420a8384 0x3fcc7950
DEBUG - timer_setfn 3fcc65f8 0x42096ee0 0x0
DEBUG - timer_setfn 3fcc660c 0x42096ed0 0x0
DEBUG - timer_disarm 3fcc7a78
DEBUG - timer_disarm 3fcc7d7c not found
DEBUG - timer_setfn 3fcc7d7c 0x420b1e08 0x0
DEBUG - timer_disarm 3fcc7a78
DEBUG - timer_disarm 3fcc7a78
DEBUG - timer_disarm 3fcc7a78
DEBUG - timer_disarm 3fcc7a78
DEBUG - Wifi mode Sta set
DEBUG - timer_disarm 3fcc7a78
INFO - wait for link
WARN - esp_wifi_internal_tx 12290
INFO - [WC] device capabilities: Ok(EnumSet(Client))
DEBUG - timer_disarm 3fcc7a78
INFO - [WC] starting..
INFO - wait for link
INFO - wait for link
INFO - wait for link
INFO - wait for link
INFO - wait for link
DEBUG - timer_disarm 3fcc7a78
```

bjoernQ, 2024-02-05:
You probably need the `phy-enable-usb` feature in `esp-wifi`: https://github.com/esp-rs/esp-wifi/blob/main/esp-wifi/README.md#usb-serial-jtag

rpisarew, 2024-02-05:
That didn't helped.

If I enable this feature my monitor output looks like that:
```
I (45) boot: ESP-IDF v5.1-beta1-378-gea5e0ff298-dirt 2nd stage bootloader
I (46) boot: compile time Jun  7 2023 08:07:32
I (47) boot: Multicore bootloader
I (51) boot: chip revision: v0.2
I (55) boot.esp32s3: Boot SPI Speed : 40MHz
I (59) boot.esp32s3: SPI Mode       : DIO
I (64) boot.esp32s3: SPI Flash Size : 16MB
I (69) boot: Enabling RNG early entropy source...
I (74) boot: Partition Table:
I (78) boot: ## Label            Usage          Type ST Offset   Length
I (85) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (93) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (100) boot:  2 factory          factory app      00 00 00010000 00fa0000
I (108) boot: End of partition table
I (112) esp_image: segment 0: paddr=00010020 vaddr=3c050020 size=10408h ( 66568) map
I (137) esp_image: segment 1: paddr=00020430 vaddr=3fc8dcd8 size=01028h (  4136) load
I (139) esp_image: segment 2: paddr=00021460 vaddr=3fcc75c0 size=00168h (   360) load
I (143) esp_image: segment 3: paddr=000215d0 vaddr=40378000 size=05cd8h ( 23768) load
I (158) esp_image: segment 4: paddr=000272b0 vaddr=00000000 size=08d68h ( 36200)
I (169) esp_image: segment 5: paddr=00030020 vaddr=42000020 size=479a8h (293288) map
I (244) boot: Loaded app from partition at offset 0x10000
I (245) boot: Disabling RNG early entropy source...
DEBUG - Creating new BME280 device at address 0x76
DEBUG - Sending soft-reset signal
```
I don't get any of my message in the code.

rpisarew, 2024-02-05:
The flash process also behaves different if I try to use the wifi controller. After the flash it just stuck with this message, until I reset the controller. I don't have to do that without wifi.

```
Chip type:         esp32s3 (revision v0.2)
Crystal frequency: 40MHz
Flash size:        16MB
Features:          WiFi, BLE
MAC address:       74:4d:bd:7f:cd:84
App/part. size:    424,432/16,384,000 bytes, 2.59%
[00:00:00] [========================================]      14/14      0x0
[00:00:00] [========================================]       1/1       0x8000
[00:00:03] [========================================]     255/255     0x10000                                    [2024-02-05T10:31:50Z INFO ] Flashing has completed!
Commands:
    CTRL+R    Reset chip
    CTRL+C    Exit

ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x15 (USB_UART_CHIP_RESET),boot:0x0 (DOWNLOAD(USB/UART0))
Saved PC:0x40378c15
0x40378c15 - core::ptr::mut_ptr::<impl *mut T>::offset
    at /Users/lowm4n/.rustup/toolchains/esp/lib/rustlib/src/rust/library/core/src/ptr/mut_ptr.rs:483
waiting for download
```

rpisarew, 2024-02-05:
Thank you very much, that solved my issue :)

closed 2024-02-05 by rpisarew