esp-rs/esp-hal #1141: Strange (out of memory?) crash

t-moe, 2024-02-06:
I have the following program, that I run on my esp32c6.

```rust
#![no_std]
#![no_main]

use embassy_sync::blocking_mutex::raw::CriticalSectionRawMutex;
use embassy_sync::pubsub::PubSubChannel;
use esp32c6_hal::{clock::ClockControl, peripherals::Peripherals, prelude::*, Delay};
use esp_backtrace as _;
use esp_println::println;

#[entry]
fn main() -> ! {
    let peripherals = Peripherals::take();
    let system = peripherals.SYSTEM.split();
    let clocks = ClockControl::max(system.clock_control).freeze();
    let mut delay = Delay::new(&clocks);

    println!("Hello world!");
    delay.delay_ms(1000u32);

    {
        println!("size of arr {}", core::mem::size_of::<[u8; 222784]>());
        let x = [0u8; 222784];
    }

    {
        // prints the same size as above
        println!("size pub sub {}", core::mem::size_of::<PubSubChannel<CriticalSectionRawMutex, [u8; 1736], 128, 1, 1>>());

        // Uncommenting the following line crashes the chip (esp-idf ?! guru meditation error: load fault)
        //let x = PubSubChannel::<CriticalSectionRawMutex, [u8; 1736], 128, 1, 1>::new();
    }

    println!("Done!");
    loop {}
}

```

Cargo.toml dependencies:
```toml
[dependencies]
esp32c6-hal = "0.8.0"
esp-backtrace = { version = "0.10.0", features = ["esp32c6", "panic-handler", "exception-handler", "print-uart"] }
esp-println = { version = "0.8.0", features = ["esp32c6"] }
embassy-sync = "0.5.0"
```

The program allocates a large structure on the stack.  
The allocation of the array (`[0u8; 222784]`) works.  
The allocation of the pub sub channel (resulting in the same allocation size), crashes the program.

Also, why do I get a guru meditation error, even if I'm **not** using esp-idf? (is this crash coming from the bootloader?)

Output, when both allocations are made:

```

[0;32mI (23) boot: ESP-IDF v5.1-beta1-378-gea5e0ff298-dirt 2nd stage bootloader[0m
[0;32mI (23) boot: compile time Jun  7 2023 08:02:08[0m
[0;32mI (24) boot: chip revision: v0.0[0m
[0;32mI (28) boot.esp32c6: SPI Speed      : 40MHz[0m
[0;32mI (33) boot.esp32c6: SPI Mode       : DIO[0m
[0;32mI (37) boot.esp32c6: SPI Flash Size : 4MB[0m
[0;32mI (42) boot: Enabling RNG early entropy source...[0m
[0;32mI (48) boot: Partition Table:[0m
[0;32mI (51) boot: ## Label            Usage          Type ST Offset   Length[0m
[0;32mI (58) boot:  0 nvs              WiFi data        01 02 00009000 00006000[0m
[0;32mI (66) boot:  1 phy_init         RF data          01 01 0000f000 00001000[0m
[0;32mI (73) boot:  2 factory          factory app      00 00 00010000 003f0000[0m
[0;32mI (81) boot: End of partition table[0m
[0;32mI (85) esp_image: segment 0: paddr=00010020 vaddr=42000020 size=37a70h (227952) map[0m
[0;32mI (140) esp_image: segment 1: paddr=00047a98 vaddr=40800000 size=00020h (    32) load[0m
[0;32mI (141) esp_image: segment 2: paddr=00047ac0 vaddr=42037ac0 size=07494h ( 29844) map[0m
[0;32mI (152) esp_image: segment 3: paddr=0004ef5c vaddr=40800020 size=01050h (  4176) load[0m
[0;32mI (155) boot: Loaded app from partition at offset 0x10000[0m
[0;32mI (160) boot: Disabling RNG early entropy source...[0m
Hello world!
size of arr 222784
size pub sub 222784
Guru Meditation Error: Core 0 panic'ed (Load access fault)
Core 0 register dump:
PC      : 0x4000179c  RA      : 0x40000000  SP      : 0x4087e610  GP      : 0x00000000
TP      : 0x00000000  T0      : 0x80824004  T1      : 0x00000001  T2      : 0x00000005
S0      : 0x00000000  S1      : 0x00000000  A0      : 0x00000000  A1      : 0x40041eb4
A2      : 0x40042064  A3      : 0x4087fcec  A4      : 0x4087fd0c  A5      : 0x40042196
A6      : 0x00000000  A7      : 0x00000066  S2      : 0x00000000  S3      : 0x00000000
S4      : 0x00000000  S5      : 0x00000000  S6      : 0x00000000  S7      : 0x00000000
S8      : 0x00000000  S9      : 0x00000000  S10     : 0x00000000  S11     : 0x00000000
T3      : 0x00000000  T4      : 0x00000000  T5      : 0x00000000  T6      : 0x00000000
MSTATUS : 0x00001881  MCAUSE  : 0x00000005  MTVAL   : 0x4004219a  INTLEVEL: 0x00000001

Stack memory:
4087e610: 0x00000001 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4087e630: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4087e650: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4087e670: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4087e690: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4087e6b0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4087e6d0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4087e6f0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4087e710: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4087e730: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4087e750: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4087e770: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4087e790: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4087e7b0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4087e7d0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4087e7f0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4087e810: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4087e830: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4087e850: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4087e870: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4087e890: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4087e8b0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4087e8d0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4087e8f0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4087e910: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4087e930: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4087e950: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4087e970: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4087e990: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4087e9b0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4087e9d0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4087e9f0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
```

It seems I'm missing something obvious here...

bjoernQ, 2024-02-06:
I can only reproduce the crash in debug mode - in release it's fine. (and usually, release should be the preferred build profile)

Seeing the Guru Meditation usually happens if something crashes before esp-hal gets a chance to set the interrupt-vector. That is not the case here - probably the code just manages to even overwrite the interrupt vector

`flip-link` is not a default - with `flip-link` enabled I get
```

Stack overflow detected at 0x42035fc6 called by 0x42033e54
0x42035fc6 - core::intrinsics::is_aligned_and_not_null
    at C:\Users\Bjoern\.rustup\toolchains\nightly-x86_64-pc-windows-msvc\lib\rustlib\src\rust\library\core\src\intrinsics.rs:2635
0x42033e54 - core::slice::raw::from_raw_parts::runtime
    at C:\Users\Bjoern\.rustup\toolchains\nightly-x86_64-pc-windows-msvc\lib\rustlib\src\rust\library\core\src\slice\raw.rs:99

```

t-moe, 2024-02-06:
Thanks for the fast reply @bjoernQ and for trying this out.

Do you think it is worth to additionally investigate why it still crashes? Or should we close this with "its up to the compiler  to allocate structures as he wants" (and two structures of the same size, dont necessarily need to be allocated the same way) ?

bjoernQ, 2024-02-06:
I guess analyzing it is a lot of effort (especially the generated assembly in debug mode is hard to read) and I'm not sure it will yield any useable insights.

If you still want to look into what is really happening and share your results - that's certainly welcome

t-moe, 2024-02-06:
Ok. Thanks.

Then I will close this, since enabling flip-link yields the behavior I expected and  I have some other pressing things to take care of first before going down the rabbit hole here.

closed 2024-02-06 by t-moe