esp-rs/esp-hal #1155: Support for psram on ESP32-PICO-V3-02 (revision v3.0)

maxwen, 2024-02-10:
Like using in Adafruit ESP32 Feather V2
I managed to make it working and testing so far showed no problems (still needs more testing though)

https://github.com/maxwen/esp-hal/commit/6ba9b84e5cbf55f4ef426de28bfa980599abfcff

Right now this is just a quick and hacky implementation for my personal projects but I would be happy to clean it up
to follow your requirements for contributing. So I was wondering how to get things started.

bjoernQ, 2024-02-12:
That's great! In general, we don't have a lot of requirements - we just have https://github.com/esp-rs/esp-hal/blob/main/CONTRIBUTING.md . Just file a PR and make sure to document what changed in CHANGELOG.md and someone will review it

sebosp, 2024-04-10:
Hi @maxwen , did you manage to create a PR to get the code upstream? I just got an M5STICKC PLUS2 and it comes with PICO V3-02 and I'd like to try esp-rs on it. Could I use your reference commit? Have you found anything that behaves strange or needs tweaking or would need testing with anything? Thanks!

PR #1424 (2024-04-22): Adapting maxwen branch to v0.16.1
maxwen@6ba9b84

closes #1155

## Thank you for your contribution!

We appreciate the time and effort you've put into this pull request.
To help us review it efficiently, please ensure you've gone through the following checklist:

### Submission Checklist 📝
- [x] I have updated existing examples or added new ones (if applicable).
- [x] My changes were added to the [`CHANGELOG.md`](https://github.com/esp-rs/esp-hal/blob/main/esp-hal/CHANGELOG.md) in the **_proper_** section.

#### Extra:
- [x] I have read the [CONTRIBUTING.md guide](https://github.com/esp-rs/esp [... cut]
commit f5dfca7f (2024-04-22): Adapting maxwen branch to v0.16.1 (#1424)

* Adapting maxwen branch to v0.16.1

maxwen@6ba9b84

closes #1155

* ESP32-PICO-V3-02 init

* rustfmt

* Update CHANGELOG.md

---------