esp-rs/esp-hal #1163: Hung up after reading from I2C

DeppLearning, 2024-02-13:
I am trying to make my boards (esp32-s3 and esp32-c6 DevKitC-1) talk to my Adafruit Atecc608A I2C device. An existing library doesn't seem to work out of the box for me, instead, [the code gets stuck on reading from I2C here](https://github.com/BlackbirdHQ/at-cryptoauth-rs/blob/46db006c2da43e82bba4508e79e9db41bf84ce94/src/datalink.rs#L122). Naturally I am trying to debug with a minimal example, which I think is this:

```rust

#![no_std]
#![no_main]

use esp32s3_hal::{clock::ClockControl, prelude::*, timer::TimerGroup, Delay, IO};
use esp_backtrace as _;
use esp_println::println;
use esp32s3_hal::i2c::I2C;

#[entry]
fn main() -> ! {
    esp_println::logger::init_logger(log::LevelFilter::Info);

    let peripherals = Peripherals::take();

    let system = peripherals.SYSTEM.split();
    
    let clocks = ClockControl::boot_defaults(system.clock_control).freeze();

    // Set up SCL & SDA
    let io = IO::new(peripherals.GPIO, peripherals.IO_MUX);
    let scl = io.pins.gpio1;
    let sda = io.pins.gpio2;

    // Construct I2C
    let mut i2c = I2C::new(peripherals.I2C0, scl, sda, 100u32.kHz(), &clocks);

    let mut delay = Delay::new(&clocks);

    for addr in 0..=127 {
        println!("Scanning Address {}", addr as u8);
    
        // Scan Address
        let res = i2c.read(addr as u8, &mut [0]);
    
        // Check and Print Result
        match res {
            Ok(_) => println!("Device Found at Address {}", addr as u8),
            Err(err) => println!("No Device Found: {:?}", err),
        }
    }

    loop {
        println!("Loop...");
        delay.delay_ms(500u32);

    }
}

```

Output of stuck program:
```
Scanning Address 0
No Device Found: AckCheckFailed
Scanning Address 1
No Device Found: AckCheckFailed
Scanning Address 2
```

I expected that it would run through the for-loop and get back with one device found. I didn't expect the reads to get hung up in any case.

I tried 
- with and without I2C device connected, same results
- with different versions of esp32s3_hal (0.14.0, 0.15.0, current main branch), same results
- on esp32s3 and esp32c6 boards
- with and without breadboard (not touching it), same results
- using different gpio pins (bunch of randoms, 1 & 2), same results
- starting the for loop at different addresses, same results
- repeatedly writing `0x00` at address `0x60` which is supposed to make the chip wake up and have it respond with some data. Same result at the read (writes don't come through either)
- setting `lto = "off"` in my cargo.toml results in my for loop getting stuck one iteration earlier
- using 50kHz I2C clock speed, it runs one iteration of the for loop less before it gets hung up
- using 2kHz or 400kHz the loop runs through, however, I am only seeing `AckCheckFailed` on all addresses including the one where I am expecting the device according to it's specs (0x60), also my original problem persists :(

(I am new to embedded, I hope this counts as minimal example, my original goal is to get this library supported on my hardware but when trying to wake up the chip, the program either get's stuck or fails here https://github.com/BlackbirdHQ/at-cryptoauth-rs/blob/46db006c2da43e82bba4508e79e9db41bf84ce94/src/datalink.rs#L122)

DeppLearning, 2024-02-15:
I tried debugging this, however, once I put log::info! in there, the program changes it's behavior (i.e. my scan loop gets NACKs on all addresses without being stuck but still I can't wake up my chip, or I'll get alternating NotAcknowledged and ExecIncomplete errors - depending on where I log) :)

By now though I am pretty sure the program gets stuck in this loop: https://github.com/esp-rs/esp-hal/blob/6e68da0f4db4a724a96f806668243cbf8828b8a0/esp-hal/src/i2c.rs#L1221-L1228

When I read out the raw interrupt status once the loop gets stuck (i.e. after 1000 iterations) I'll get this:

```
1000001010000010

--------------> I2C_TXFIFO_WM_INT 
--------> TRANS_COMPLETE_INT_RAW_R
------> TRANS_START_INT_RAW_R
> DET_START_INT_RAW_R
```

So if I understand this correctly, there is no I2C_END_DETECT_INT bit set. Does this mean the program is waiting for more bits to be set in this register but that never happens? When I reset (like the check_errors method does on errors) and error out from that loop (i.e. after 1000 iterations), it will get a NACK on the next address scanned, but again, after that it'll end up stuck in the same loop.

bjoernQ, 2024-02-16:
I guess resetting after running into an error is the right thing to do since the peripheral won't be able to recover otherwise. Probably what the `reset` function does currently is just not enough.

Probably makes sense to check what esp-idf does, e.g.
https://github.com/espressif/esp-idf/blob/master/components/hal/esp32c3/include/hal/i2c_ll.h
https://github.com/espressif/esp-idf/blob/master/components/driver/i2c/i2c.c

DeppLearning, 2024-02-16:
Can you confirm this is a bug? 

> I guess resetting after running into an error is the right thing to do since the peripheral won't be able to recover otherwise. Probably what the `reset` function does currently is just not enough. [...]

Regarding the reset, they seem to handle the case where dead slave devices hold SDA low, which I haven't seen in this crate: https://github.com/espressif/esp-idf/blob/4d90eedb6ef32ff949c6f298a3d845f9f07bac92/components/driver/i2c/i2c.c#L622 . However, I am getting stuck in the loop without any slaves connected, so probably not it? 

Its hard to tell for me what they are really doing for reset, I'll take some more time to look at it to try to parse it and will come back if I find anything (I deem) interesting.

I also checked their check_errors equivalent (?) and spotted some differences: 

This looks like they have another error condition entirely that might apply:
https://github.com/espressif/esp-idf/blob/4d90eedb6ef32ff949c6f298a3d845f9f07bac92/components/driver/i2c/i2c.c#L561

And also they seem to throw timeout errors on more conditions than this crate does
https://github.com/espressif/esp-idf/blob/4d90eedb6ef32ff949c6f298a3d845f9f07bac92/components/hal/esp32c3/include/hal/i2c_ll.h#L959

Then there is this comment https://github.com/espressif/esp-idf/blob/4d90eedb6ef32ff949c6f298a3d845f9f07bac92/components/driver/i2c/i2c.c#L532 , which doesn't seem to be handled either by this crate. However, if anything I should get more interrupts rather than fewer, so it's probably not it either?

bjoernQ, 2024-02-19:
Interestingly I ran your example in debug mode on that made it work. Since it shows the same behavior on C6 in release mode I don't think there is anything triggering a mis-compilation.

Probably it works in debug mode since the code runs much, much slower - so it would be a timing related problem

DeppLearning, 2024-02-19:
Oh boy I found it, I switched up sda and scl in this line 

```rust
 let mut i2c = I2C::new(peripherals.I2C0, scl, sda, 100u32.kHz(), &clocks); 
```

it should be:

```rust
 let mut i2c = I2C::new(peripherals.I2C0, sda, scl, 100u32.kHz(), &clocks); 
```

sorry for the fuzz, username checks out I guess :)

closed 2024-02-19 by DeppLearning

MabezDev, 2024-02-19:
> Oh boy I found it, I switched up sda and scl in this line
>
> ```rust
>  let mut i2c = I2C::new(peripherals.I2C0, scl, sda, 100u32.kHz(), &clocks);
> ```
>
> it should be:
>
> ```rust
>  let mut i2c = [...]

Hehe, it's a rite of passage to swap pins as an embedded developer :D

DeppLearning, 2024-02-19:
> Hehe, it's a rite of passage to swap pins as an embedded developer :D

however... switching the pins solves my problem, can also wake up my chip now. But. Why does the initial example I showed is affected by this? For that example, the pins used shouldn't matter (since they're not connected anyways?)

bjoernQ, 2024-02-19:
I guess since you got your wiring correct now it works fine - the problem starts when we don't get NACKs. I will create a PR for that problem (which apparently isn't really yours)

DeppLearning, 2024-02-19:
> I guess since you got your wiring correct now it works fine - the problem starts when we don't get NACKs. I will create a PR for that problem (which apparently isn't really yours)

I was a bit too quick, when I remove log lines from the esp code, the problem appears again :) at least I can workaround the issue now

bjoernQ, 2024-02-19:
With #1184 you should be able to use it without the log statements