esp-rs/esp-hal #1171: Multi-core critical-section is not sound(?)

i404788, 2024-02-15:
I'm currently debugging a weird issue with embassy multi-core related to critical section, as it seems like it tries to double-unlock the `MULTICORE_LOCK` from different threads when using the embassy executor and I'm not entirely sure why it happens.

What seems to happen:
- esp-hal's embassy impl uses `set_alarm` for the executor poll (Seems to be on thread_id=1, token=4294967295)
- In `set_alarm` it aquires a `critical_section`, and at the end attempts to drop it
- *While Dropping* (at https://github.com/esp-rs/esp-hal/blob/main/esp-hal/src/lib.rs#L279) a level3 interrupt hits
- The interrupt gets handled by `time_driver_timg.rs` `EmbassyTimer::on_interrupt`
- `EmbassyTimer::on_interrupt` aquires it's own `critical_section` and when it tries to drop the owner isn't `thread_id()` (thread_id seems to be 0 at this stack frame).

I'm not sure how it's possible to have 2 threads in one call stack, but it seems like thread_1 unlocks, and before finishing the sync thread_0 locks, causing thread_1 to overwrite the owner. Let me know if I'm missing something

MabezDev, 2024-02-15:
@bugadani sorry for the ping, but do you have any ideas on this one?

bjoernQ, 2024-02-16:
Oh this 
```rust
fn get_raw_core() -> usize {
    ((xtensa_lx::get_processor_id() as usize & 0x2000) != 0) as usize
}
```

will very likely produce a value of 1

MabezDev, 2024-02-16:
Git blaming that line points me to this PR: https://github.com/esp-rs/esp-hal/pull/1126 I'm not sure what the crash was? I don't see an issue associated with it.

bjoernQ, 2024-02-16:
IIRC there was a discussion in the Matrix channel - ah yes ... there it is: https://matrix.to/#/!LdaNPfUfvefOLewEIM:matrix.org/$LXNqhvjxecSQlJFK-mmJCIYYJ_fgStulDsr5r1mNb-o?via=matrix.org&via=tchncs.de&via=mozilla.org

MabezDev, 2024-02-16:
@i404788 Apologies for the breakage, I have the fix in https://github.com/esp-rs/esp-hal/pull/1175, would you mind giving it a try? Likewise @ProfFan would you mind testing that I haven't broken your use case too?

ProfFan, 2024-02-16:
> Git blaming that line points me to this PR: #1126 I'm not sure what the crash was? I don't see an issue associated with it.

Oops, it's me :P

closed 2024-02-19 by MabezDev