esp-rs/esp-hal #1188: Are `GpioPin` type state generics actually helpful?

MabezDev, 2024-02-20:
We have a lot of type information stored with our GPIO types, and I'm wondering if it's valuable to have or not. The traditional use case for typestates is to ensure that specific pins that have a specific purpose are only used for that purpose _and_ are in the correct state for that purpose.

For esp's we are gifted with the luxury of the IO mux, where any<sup>1</sup> pin can be used for any purpose, so the first case is covered. I took a look at our peripheral driver initialization operations, _none_ of them take a `GpioPin<TYPE>` directly, instead we take the raw pin and configure it in the driver. (I would also like to note at this point, I love this about esp-hal, I think it's one of the nicest features that drivers take care of that!)  This covers the second case from above. This begs the question, are the type state generics actually helpful?

I think at this point I would have to say no, but I might be missing something. I would love to hear some alternative points.

If we did choose to remove it, it would _massively_ simplify the GPIO module. There would be just one parameter on `GpioPin` which would be the number, and `AnyPin` would then be truly generic free.

<sup>1</sup>: There are some limitations around USB pins.

jessebraham, 2024-02-20:
I do not think the generics are especially useful (and get in the way more than anything). I've been of the opinion that our GPIO driver needs a full redesign for quite some time now, so I'm in favour of eliminating as much of this as possible.

bjoernQ, 2024-02-21:
After thinking about it for some time I'd say no.

>  I've been of the opinion that our GPIO driver needs a full redesign for quite some time now, so I'm in favour of eliminating as much of this as possible.

Maybe we can avoid doing a rewrite and get closer to what you want with a refactoring resulting from this? I guess this will be quite a big refactoring task and improving things incrementally is probably a good way forward. Did you outline the ideas somewhere? Just to make sure things don't go into opposite directions

Dominaezzz, 2024-03-08:
> but I might be missing something

I think you're forgetting about e-hal drivers. Some LED or button driver will want a gpio pin specifically in the input or output state.

MabezDev, 2024-03-09:
> I think you're forgetting about e-hal drivers. Some LED or button driver will want a gpio pin specifically in the input or output state.

Most of our pins implement both traits anyway. There are only a handful of pins that cannot be used as inputs and we just wouldn't implement the trait on those pins. The typestate information isn't really that relevant to what traits they implement.

MabezDev, 2024-03-09:
Ah, I see what you mean (sorry it's late here), the pins still need to be put into input or output mode. I guess one level of typestate could be okay. Or we could go down the path of embassy, and esp-idf-hal where the mode is a struct which takes a pin i.e `Input::new(io.gpio0)`.

My main reason for opening this issue is that we have types like `Gpio0<Input<PullDown>>`, not unwieldy, but does `PullDown` actually help in anyway or is it just noise?

bjoernQ, 2024-03-11:
I think having state for Input, Output and OpenDrainOutput is something we'll need.  I see no real use in the second level of typestate - I guess having separate structs or a state on just one struct should be equivalent but given there is a "trend" to have separate structs we might want that 🤔 but no hard-opinion on that from my side. Introducing separate structs might be a more drastic change but if we want to have it it's probably better to do that in one (bigger) change

Zagitta, 2024-03-20:
I've run into this and would agree that the type states get in the way when using peripherals at least.

For example, I've been trying to create an abstraction over the `mcpwm` peripheral(s) which should take 12 output-capable pins and pass them into `Operator::with_pins`.
This is fine and dandy in normal non-embassy land because the abstraction could be made generic over the pins  (despite that being a big mess of repeated generics), however with embassy it just plainly isn't possible because tasks can't be generic so I would have to spell out the specific pins in the task parameter defeating the point.

More concretely this problem arises because [`Operator::with_pins`](https://github.com/esp-rs/esp-hal/blob/1f155cf301cd35b0ab77fff362c84dc953b39eda/esp-hal/src/mcpwm/operator.rs#L282) has the bound `impl Peripheral<P = PinA> where PinA: OutputPin` but `OutputPin` is for `AnyPin` like this: `impl<MODE, TYPE> OutputPin for AnyPin<MODE, TYPE> where MODE: OutputMode, TYPE: IsOutputPin` but `AnyPin` has `TYPE = ()` which naturally doesn't implement `IsOutputPin`.
This in turn means that my abstraction either has to chose between `InputOutputPinType` or `InputOutputAnalogPinType` because that's the two types that implement `IsOutputPin`.

It's not really possible to work around this either with TAIT (as far as I understand) because I can't resolve the `TYPE` to concrete types in the abstraction module without limiting which pins goes to which pwm output.