esp-rs/esp-hal #1191: Support embassy-usb on ESP32-S2 and ESP32-S3

Dominaezzz, 2024-02-20:
Whilst this crate supports `usb-device` (which is not `async`), it'd be really cool if it also supported the async `embassy-usb`.

Some efforts have begun [here](https://github.com/embassy-rs/embassy/issues/2494).

Dominaezzz, 2024-03-08:
These registers should be accessible by index. The 0 at the end can vary from 0 to 6.
```
pub type DOEPCTL0 = crate::Reg<doepctl0::DOEPCTL0_SPEC>;
pub type DOEPINT0 = crate::Reg<doepint0::DOEPINT0_SPEC>;
pub type DOEPTSIZ0 = crate::Reg<doeptsiz0::DOEPTSIZ0_SPEC>;
pub type DOEPDMA0 = crate::Reg<doepdma0::DOEPDMA0_SPEC>;
pub type DOEPDMAB0 = crate::Reg<doepdmab0::DOEPDMAB0_SPEC>;
```

bjoernQ, 2024-03-11:
Probably the PACs should get patched to collect these registers into arrays. We use https://github.com/rust-embedded/svdtools for patching SVDs in https://github.com/esp-rs/esp-pacs

PR #1517 (2024-05-01): embassy-usb support
## Thank you for your contribution!

We appreciate the time and effort you've put into this pull request.
To help us review it efficiently, please ensure you've gone through the following checklist:

### Submission Checklist 📝
- [x] I have updated existing examples or added new ones (if applicable).
- [x] My changes were added to the [`CHANGELOG.md`](https://github.com/esp-rs/esp-hal/blob/main/esp-hal/CHANGELOG.md) in the **_proper_** section.
#### Extra:
- [x] I have read the [CONTRIBUTING.md guide](https://github.com/esp-rs/esp-hal/blob/main/CONTRIBUTING.md) and followed  [... cut]
closed 2024-05-01 by jessebraham