esp-rs/esp-hal #1194: Refactor reg_X_bit functions

MabezDev, 2024-02-21:
We currently have duplicates of these reg functions where the svd's don't have enough info, and we have to resort back to raw register writes. E.g: https://github.com/search?q=repo%3Aesp-rs%2Fesp-hal%20reg_set&type=code.

We should cleanup and modularise their usage/definitions across our chip line-up.

cc: @playfulFence who initially noticed the issue.

bjoernQ, 2024-02-21:
Maybe we could even avoid those functions and use something like https://crates.io/crates/tock-registers

But I think such code originates from 1:1 translation of esp-idf code - but once such code is known working, we could improve it

jessebraham, 2024-02-21:
Maybe I'm missing something, but why are we going to all this effort instead of just fixing the PACs?

bjoernQ, 2024-02-21:
> Maybe I'm missing something, but why are we going to all this effort instead of just fixing the PACs?

That would certainly be the best solution for registers! I somehow thought of I2C_BBPLL which we also deal with in those places

MabezDev, 2024-02-21:
Good idea! Let's do that instead, I've updated the title.

jessebraham, 2024-02-21:
I don't think I will be able to do this any time soon unfortunately, but if somebody else wants to tackle it I can at least explain what needs to be done to them, so just lemme know. If nobody has tackled this in a month or two when I wrap up my currently projects then I will take care of it at that point.

playfulFence, 2024-07-10:
Just to have some part of an investigation here: 
Most of the cases of manipulations with raw register addresses are related to `regi2c_*` functions, which has a tracking issue here https://github.com/esp-rs/esp-hal/issues/1740. Which means that this issue highly depends on it. At this point, I think, I will temporarily switch to analyzing `regi2c_*` functions.

playfulFence, 2024-08-15:
Work has been done on this issue (see in PRs above). At the moment, most operations with raw register addresses are performed in `regi2c_*` functions. Due to the complexity and black-box nature of this part, it was decided to postpone this work, so I consider this issue closed.
I will keep it open until the end of this day and, if there are no hard opinions, I will close it as successfully completed

MabezDev, 2024-08-15:
We can track the regi2c stuff in #1740, so I think we can close this now.

closed 2024-08-15 by MabezDev