esp-rs/esp-hal #1199: Remove third-party re-exports in prelude

MabezDev, 2024-02-24:
With e-h this is particularly annoying as eh0.2, eh1 and eha have some traits with overlapping methods, namely delay_ms. This makes it impossible to call the right delay without using the Fully Qualified Syntax.

Ben-PH, 2024-02-26:
I'd like to be able to make a cross platform app that takes in values constrained by implementors of embedded-hal traits. EG:

```
#![no_std]
#![no_main]

use embedded_hal::{digital::OutputPin, delay::DelayNs};

pub fn cross_blinky<P: OutputPin, T: DelayNs>(mut led_pin: P, mut timer: T) -> ! {
    loop {
        led_pin.set_high().unwrap();
        timer.delay_ms(500);
        led_pin.set_low().unwrap();
        timer.delay_ns(500);
    }

}
```

...but it seems that this cannot be done with esp32c3_hal, for example. it implements its own re-export of OutputPin, so in order to pass in an esp32c3 pin, I would need to make a wrapper that implements the wrapped gpio pin. I tried it, and it's not straight forward.

I'll get started on this now.

jessebraham, 2024-02-26:
Duplicate of #849

closed 2024-02-26 by jessebraham