esp-rs/esp-hal #1203: Double check results from calibrated ADC readings

bjoernQ, 2024-02-26:
At least on C6 we see suspicious results.

e.g. for AdcCalCurve
```
PIN2 ADC reading = 7734 mV
PIN2 ADC reading = 6681 mV
PIN2 ADC reading = 6832 mV
PIN2 ADC reading = 6992 mV
```

for AdcCalLine
```
PIN2 ADC reading = 7985 mV
PIN2 ADC reading = 7343 mV
PIN2 ADC reading = 7506 mV
PIN2 ADC reading = 7007 mV
PIN2 ADC reading = 7017 mV
PIN2 ADC reading = 7506 mV
```

AdcCalBasic
```
PIN2 ADC reading = 8223 mV
PIN2 ADC reading = 7218 mV
PIN2 ADC reading = 7385 mV
PIN2 ADC reading = 7553 mV
PIN2 ADC reading = 7207 mV
PIN2 ADC reading = 7385 mV
```

jessebraham, 2024-03-26:
Not sure what's going on here, but based on my testing C6 worked without issue. I did see incorrect readings for the C3 and S3, however.

Possibly due to different chip revisions or something? Not sure, needs further investigating. I did double-check the curve coefficients at the very least and they seem to be more or less correct.

jessebraham, 2024-06-10:
I've just done a quick round of testing on all devkits I have available to me. I've included the revisions as reported by `espflash board-info` as this may make a difference.

Anyway, results are all over the place, so we need to spend some time fixing all this for sure.

|      Chip       | None (GND/VCC) | Basic (GND/VCC) | Line (GND/VCC) | Curve (GND/VCC) |
| :-------------: | :------------: | :-------------: | :------------: | :-------------: |
|  ESP32 (v3.0)   |   4095/4095    |     &mdash;     |    &mdash;     |     &mdash;     |
| ESP32-C2 (v1.0) |   2138/4095    |     0/3450      |     0/3343     |     &mdash;     |
| ESP32-C3 (v0.3) |   1690/4095    |     7/4095      |     5/2960     |     6/2860      |
| ESP32-C6 (v0.0) |   2160/4081    |     1/3393      |     0/3262     |     0/3250      |
| ESP32-H2 (v0.0) |   2160/4081    |     &mdash;     |    &mdash;     |     &mdash;     |
| ESP32-S2 (v0.0) |   8192/3470    |     &mdash;     |    &mdash;     |     &mdash;     |
| ESP32-S3 (v0.1) |   1960/4095    |     0/4096      |     0/3456     |     0/64562     |

I know it's a bit of a pain, but if 1 or 2 other people would be able to do this same test that would be helpful. More data won't hurt here.

bjoernQ, 2024-06-11:
|      Chip       | None (GND/3V3) | Basic (GND/3V3) | Line (GND/3V3) | Curve (GND/3V3) |
| :-------------: | :------------: | :-------------: | :------------: | :-------------: |
|  ESP32 (v1.0)   |   0/4095       |     &mdash;     |    &mdash;     |     &mdash;     |
| ESP32-C2 (v2.0) |   2155/4095    |     0/3460      |     0/3308     |     &mdash;     |
| ESP32-C3 (v0.3) |   1674/4095    |     10/4095     |     10/2885    |     13/2796     |
| ESP32-C6 (v0.0) |   2145/4081    |     1/3377      |     5/35367    |     0/34789     |
| ESP32-H2 (v0.1) |   2144/4081    |     &mdash;     |    &mdash;     |     &mdash;     |
| ESP32-S2 (v0.0) |   2825/8191    |     &mdash;     |    &mdash;     |     &mdash;     |
| ESP32-S3 (v0.1) |   1971/4095    |     0/4095      |     0/3540     |     0/3159      |

* ESP32: the example should use GPIO32 but I only get readings when using GPIO33

JurajSadel, 2024-06-11:
|      Chip       | None (GND/3V3) | Basic (GND/3V3) | Line (GND/3V3) | Curve (GND/3V3) |
| :-------------: | :------------: | :-------------: | :------------: | :-------------: |
|  ESP32 (v3.0)   |   3830/4095    |     &mdash;     |    &mdash;     |     &mdash;     |
| ESP32-C2 (v2.0) |   2180/4095    |     1/3431      |     1/3374     |     &mdash;     |
| ESP32-C3 (v0.3) |   1693/4095    |     3/4095      |     3/2949     |     3/2852      |
| ESP32-C6 (v0.0) |   2160/4081    |     0/3313      |     0/3295     |     0/3283      |
| ESP32-H2 (v0.2) |   2160/4081    |     &mdash;     |    &mdash;     |     &mdash;     |
| ESP32-S2 (v0.0) |   2821/8191    |     &mdash;     |    &mdash;     |     &mdash;     |

I don't have S3 right now.

lure23, 2024-06-12:
My results with `adc_cal` example from `esp-hal` ([`d2a9389`](https://github.com/esp-rs/esp-hal/tree/d2a93894f556af948e37951b4b9433ee61654f34)):

|Board|Chip|None|Basic|Line|Curve|
|---|---|---|---|---|---|
|ESP32-C3-DevKitC-02 "1.1"|ESP32-C3 (v0.4)|1680..1689 / 4095|0 / 4095|0..1 / 3014|1..2 / 2906|
|ESP32-C3-DevKit-RUST-1 "1.2a 04/2022"|ESP32-C3 (v0.4)|1680..1690 / 4095|0..18 / 4095|5..17 / 3040|6..14 / 2927|
|ESP32-C6-DevKitM-1 "1.0"|ESP32-C6 (v0.0)|2176..2177 / 4081|0..1 / 3281..3297|0..1 / 3304..3321|0..1 / 3292..3309|

Values "gnd / 3v3" like from the other people. I don't know what to make of this. Hope it helps.

The quoted strings are what's seen on the underside of the dev board PCB. DevKitC was controlled by uart, others by JTAG (don't think that matters).

jessebraham, 2024-06-12:
Thanks everybody for the additional info. I will start digging into this next week most likely.

sourcebox, 2024-06-24:
I did some more detailled measurements with my ESP32-S3-DevKitC-1. I connected a pot to GPIO3 and set some inbetween voltages. As you can see, all calibrated results below 0.3V are near 0, so there's a large gap, while the raw values are still changing. Above 3.0V, the curve calibration is completely broken.

| Voltage   | None      | Basic     | Line      | Curve     |
|-----------|-----------|-----------|-----------|-----------|
|     0.0   | 1973      |     0     |     0     |     0     |
|     0.1   | 2095      |     0     |     0     |     0     |
|     0.2   | 2233      |     0     |     0     |     0     |
|     0.3   | 2348      |    93     |    83     |    83     |
|     0.4   | 2473      |   222     |   188     |   195     |
|     0.5   | 2582      |   331     |   275     |   285     |
|     0.6   | 2711      |   463     |   385     |   393     |
|     0.7   | 2857      |   614     |   512     |   518     |
|     0.8   | 2957      |   717     |   585     |   597     |
|     0.9   | 3079      |   829     |   689     |   694     |
|     1.0   | 3207      |   955     |   805     |   794     |
|     1.5   | 3870      |  1628     |  1349     |  1323     |
|     2.0   | 4095      |  2180     |  1830     |  1785     |
|     2.5   | 4095      |  2967     |  2465     |  2404     |
|     3.0   | 4095      |  3800     |  3172     | 64400     |
|     3.2   | 4095      |  4071     |  3395     | 65534     |

After doing these measurements, I did a quick check against the C example from the IDF and it shows that calibrated results start at about 25mV of input voltage. They are not very accurate, but they are not zeroed.

dimpolo, 2025-01-16:
@jessebraham did you manage to make some progress?

Lenoirio, 2025-04-17:
I was also fighting with this issue the whole day because I wanted to get a thermistor working with an ESP32-S3.  From what I measured at the GPIO (MultiMeter) and the absolute ADC-readings it seems that it's using a 13bit-resolution (that's the default from what I saw in the source). 
But the values are capped at 12bit. E.g. 4095 was the highest value I could get from the oneshot_read and thus all values higher than about  1.6V were just reported as 4095

I solved it by using a 33k resistor in the voltage-divider and thus shifting the values of the thermistor in a range below that threshold. Now I can measure the values in the range I'm interested in.

jessebraham, 2025-04-22:
Apologies for the lack of updates. We had decided that ADC was not a priority, and as such I spent my time elsewhere, so have not really made any progress on this front.

However, recently I discovered that the eFuse field definitions we parsed to generate each device's `efuse` module contain a number of errors, and have done a fair bit of work in `espflash` correcting this. This week I will port these changes over to `esp-hal`, and hopefully this will improve calibration given that we are reading certain values from eFuse fields where, in the HAL's current state, are probably incorrect.

I will post any updates here once I have completed this work. No guarantees it will resolve the issues, but I'm hoping it does.

mikabytes, 2025-10-16:
Tried again with rc.1 but it didn't show significant changes for me. However, switching from `cfg.enable_pin` to `cfg.enable_pin_with_cal::<_, AdcCalLine<_>>` was significant. Pins now show 0 when pulled to GND. So maybe what I was seeing was just some calibration done by IDF that was lingering around somehow. I'm sort of new to ESP programming so it's probably my mistake.

dimpolo, 2025-11-17:
```
> espefuse --port COM7 dump
espefuse v5.1.0
Connecting...
Detecting chip type... ESP32-S3

=== Run "dump" command ===
BLOCK0          (                ) [0 ] dump: 00000000 00000000 00000000 00000000 00000000 00000000
MAC_SPI_8M_0    (BLOCK1          ) [1 ] dump: 4342da80 000048ca 00000000 03080000 3cd8b000 7800dccc
BLOCK_SYS_DATA  (BLOCK2          ) [2 ] dump: bcb4baa7 8b04eb96 2d2e09f5 1557a8e4 cf2a9531 5453c4c4 acbebae2 447cfcc2
...
```

user12257, 2026-02-21:
Hi, 
I am having accuracy Issues on an esp32c6 with every type of calibration. Is there any progress or maybe something I could do to help (as a beginner in Rust and embedded)?

closed 2026-09-25 by bjoernQ