esp-rs/esp-hal #1205: Documentation of esp-hal

MabezDev, 2024-02-26:
## lib docs

- [ ] Avoid documenting features that don't apply to the current chip
- [ ] `cfg` away (or add) chip specific documentation for drivers

## external

- [ ] Book updates
- [ ] Template updates
- [ ] Migration guide/post?

JurajSadel, 2024-02-27:
I'm going to start with READMEs

jessebraham, 2024-02-28:
I've moved some of the documentation tasks to a separate issue: https://github.com/esp-rs/esp-hal/issues/1221

jessebraham, 2024-03-04:
Fleshed out `examples/README.md` in #1237

jessebraham, 2024-03-04:
READMEs have been updated.

jessebraham, 2024-03-05:
@MabezDev @bjoernQ what are your thoughts on delaying the remaining two points here, in the interest of getting the release out? I think it's fine if we tackle them both post-release, personally.

closed 2024-03-05 by MabezDev