esp-rs/esp-hal #1207: ESP32-C3: TWAI broken

bjoernQ, 2024-02-27:
While testing #1196 it was found that TWAI is apparently broken for ESP32-C3 while it's still working for ESP32-S3

bjoernQ, 2024-02-29:
After a lot of fiddling, it seems that for whatever reason things don't work for me when using GPIO 1 as RX.
Using GPIO 2 however works fine apparently.  Would be good if anyone else could check if GPIO1 is fine for them

msvisser, 2024-08-13:
I have been playing with Rust on the ESP32-C3 and I ran into the same problem where GPIO1 does not work as RX for the TWAI peripheral. I already had some code written in C using the ESP-IDF which did work on my hardware, so I knew this was a software issue.

To determine where the issue is, I started comparing the registers of the peripherals after they were initialized by C and by Rust. I noticed that the `TWAI_CLOCK_OFF` bit in the `TWAI_CLOCK_DIVIDER_REG` was not set by `esp-hal` but was set by the ESP-IDF. I then added some code to `TwaiConfiguration::new_internal` to set this bit, and that fixed the issue for me!

```rust
// Disable CLKOUT
T::register_block()
    .clock_divider()
    .write(|w| w.clock_off().bit(true));
```

I will try to open a pull request tomorrow to incorporate this change and fix the issue.

ProfFan, 2024-08-15:
@msvisser Is this still the case after #1906 is merged? In theory the choice of which GPIO should not matter at all as they all go through the GPIO matrix.

msvisser, 2024-08-16:
Okay I have checked again, and it seems like #1906 did fix the problem. I was tripped up by #1929, because before it was merged the peripheral was always initialized in listen only mode, while I was trying to transmit first before receiving. When I forked the repo to implement my fix it suddenly started working because I was using a newer version of `esp-hal`.

closed 2024-08-16 by MabezDev