esp-rs/esp-hal #1216: Uart: polarity inversion

anti-social, 2024-02-28:
As I can see an Esp32c3 chip supports inversion: [uart_set_line_inverse](https://docs.espressif.com/projects/esp-idf/en/stable/esp32c3/api-reference/peripherals/uart.html#_CPPv421uart_set_line_inverse11uart_port_t8uint32_t)

Have tested it from Micropython connecting to external module pins of my Radiomaster handset (it uses half-duplex serial interface with inverted polarity). So it would be nice to have the option in Rust.

bjoernQ, 2024-02-28:
There is an open PR for that already here:  https://github.com/esp-rs/esp-hal/pull/1151

But it _should_ already work via a slightly different mechanism introduced in https://github.com/esp-rs/esp-hal/pull/912

anti-social, 2024-02-28:
Oh, sorry. Haven't searched in PRs.

> But it should already work via a slightly different mechanism introduced in https://github.com/esp-rs/esp-hal/pull/912

To make it work I should configure inversion on uart pins and then create a uart device?

bjoernQ, 2024-02-28:
> Oh, sorry. Haven't searched in PRs.

No problem! 😄 

> To make it work I should configure inversion on uart pins and then create a uart device?

Yes, it should work that way but t.b.h. I haven't tried it with UART myself, yet

jessebraham, 2024-06-25:
As discussed in #1703, it appears we already have a method of inverting the GPIO, which are then in turn correctly configured for inverted UART. As such, perhaps no additional code is required, just some updates to documentation.

closed 2024-06-27 by jessebraham