esp-rs/esp-hal #1233: Embassy main proc_macro doesn't have rustdoc for every generated element

AnthonyGrondin, 2024-03-03:
The procedural macro to generate the `main` for embassy doesn't have rustdoc for every generated element.
The following expansion shows that some members have neither documentation or `#[doc(hidden)]` making them discoverable by rustdoc making the doc coverage fail.

```rust
#[doc(hidden)]
async fn ____embassy_main_task(spawner: Spawner) {
    { ::core::panicking::panic("not yet implemented") }
}
fn __embassy_main(spawner: Spawner) -> ::embassy_executor::SpawnToken<impl Sized> {
    type Fut = impl ::core::future::Future + 'static;
    const POOL_SIZE: usize = 1;
    static POOL: ::embassy_executor::raw::TaskPool<Fut, POOL_SIZE> = ::embassy_executor::raw::TaskPool::new();
    unsafe { POOL._spawn_async_fn(move || ____embassy_main_task(spawner)) }
}
unsafe fn __make_static<T>(t: &mut T) -> &'static mut T {
    ::core::mem::transmute(t)
}
#[doc(hidden)]
#[export_name = "main"]
pub unsafe extern "C" fn __xtensa_lx_rt_main_trampoline() {
    __xtensa_lx_rt_main()
}
#[allow(clippy::inline_always)]
#[inline(always)]
fn __xtensa_lx_rt_main() -> ! {
    let mut executor = esp32s3_hal::embassy::executor::Executor::new();
    let executor = unsafe { __make_static(&mut executor) };
    executor
        .run(|spawner| {
            spawner.must_spawn(__embassy_main(spawner));
        })
}
```

jessebraham, 2024-03-03:
There are a large number of documentation improvements which are still needed, which you can track in #1221. This is all being worked on and should be much better by the time of our next release.

I will add this to the list.

AnthonyGrondin, 2024-03-05:
Thank you very much for taking care of this.

After further investigation, this is a general proc_macro issue and how they work in Rust. Since they get treated as user code, they get picked up by `rustdoc` and `cargo doc` but no warning is given with `#[warn(missing_docs)]`.

The good news is that this "issue" doesn't affect all code but only proc_macro code that gets expanded into user's code. A few `#[doc ...]` should do the trick.

closed 2024-03-21 by jessebraham