esp-rs/esp-hal #1245: Remove `'static` bound requirement on DMA buffers?

MabezDev, 2024-03-06:
To be 100% safe, buffers used with the embedded-dma traits _must_ be static. This is because `core::mem::forget` is safe in Rust, and if a user were to forget a driver whilst a DMA transaction was occurring, the buffer could go out of scope and UB would occur. Not good.

The issue is, is that `'static` bounds are a pain to work with, and I think in practice it's unlikely for users to `forget` our drivers. I think it might be a good idea to have a trade-off here, to remove the `'static` bounds to make the drivers easier to use. We can document each driver, and even have a section in our top level docs mentioning our drivers are _not_ leak (forget) safe and must be dropped properly or manually stopped before `forget`ing.

MabezDev, 2024-03-11:
When we look into this, we should also fix #954 as removing the 'static bound will make it more likely that the buffer could be in cache (which the DMA can't write to).

bjoernQ, 2024-07-16:
Maybe we should have our own type for buffers which implements `Drop` and suggest to deny https://rust-lang.github.io/rust-clippy/v0.0.212/index.html#mem_forget

I haven't tried it - just an idea after sleeping over this

PR #1837 (2024-07-25): Lift `static` requirement on DMA buffers
## Thank you for your contribution!

We appreciate the time and effort you've put into this pull request.
To help us review it efficiently, please ensure you've gone through the following checklist:

### Submission Checklist 📝
- [ ] I have updated existing examples or added new ones (if applicable).
- [x] I have used `cargo xtask fmt-packages` command to ensure that all changed code is formatted correctly.
- [x] My changes were added to the [`CHANGELOG.md`](https://github.com/esp-rs/esp-hal/blob/main/esp-hal/CHANGELOG.md) in the **_proper_** section.
-  [... cut]
closed 2024-07-25 by MabezDev